zephyrproject-rtos/zephyr #22293: Bluetooth: controller: Adv Data buffer management design

cvinayak, 2020-01-29:
- Primary channel double buffering
- Secondary channel double buffering
- Secondary channel chaining 
- Secondary channel chain double buffering
- Buffer allocation with waiting for availability
-- Adv Data set in thread mode, use OS features to synchronize free buffer availability

lukasbrchl, 2021-08-22:
Hello @cvinayak,

In our application, we have come across some weird behavior related to the reliability of Bluetooth Legacy and Long Range advertisements. We noticed that if we schedule one BT4 and one BT5 LR advertising to the same frequency (e.g. 200 ms), then only one of them is mostly transmitted. The same applies when the frequencies are not the same but close to each other (e.g. 200ms and 220 ms). This is quite a pity if you have more advertising sets because the multiples of frequencies are often the same. We verified this behavior on BT Sniffer ([Sniffle](https://github.com/nccgroup/Sniffle)).

My question is, could this issue be caused by buffering you are discussing in this issue? Do you plan to work on the buffering somewhen in near future? We would like to help, but the host & controller implementation is too complex for our understanding so the best we could do is help with testing/verifying.

Thanks, Lukas

cvinayak, 2021-08-23:
Hi @lukasbrchl, Legacy (BT4) and Extended (BT5) advertising radio events use random delay of upto 10 ms alongwith your applications requested 200 ms. They should be each drift apart on every interval.

> My question is, could this issue be caused by buffering you are discussing in this issue? Do you plan to work on the buffering somewhen in near future? 

No, data buffering has no relation to radio event scheduling. Data buffering is work in progress, I have updated the description.

> We would like to help, but the host & controller implementation is too complex for our understanding so the best we could do is help with testing/verifying.

You can start with sending a PR with a simple application sample close to your requirement, which I can use to reproduce the issue you are facing. Subsequently, this sample can be merged upstream if its useful for general user to demonstrate a feature.

carlescufi, 2021-12-22:
Done

closed 2021-12-22 by carlescufi