zephyrproject-rtos/zephyr #22301: k_msgq_put() semantics definition

chrisstjohn, 2020-01-29:
**Is your enhancement proposal related to a problem? Please describe.**
The documentation doesn't contain a precise description of the handling of data by k_msgq_put(). This is particularly important when used from an ISR context.

**Describe the solution you'd like**
Doxygen documentation amended to read something like "the message content is copied to the message queue and the data pointer is not retained" and the prototype changed to use const void* underlining that the message content cannot be modified as a result of this call.

**Describe alternatives you've considered**
I resorted to reading the source code and casting away const.

**Additional context**
The structured message queue mechanism is elegant and allows providers to, for instance, construct messages on their stack before pushing them to the queue. This appears to be understood by experienced contributors (see subsys/shell/shell_log_backend.c) but is not at all clear to novices.

carlescufi, 2020-01-29:
@chrisstjohn would you consider sending a Pull Request with a proposal?

XavierChapron, 2020-06-11:
I've also been suprised that current prototype is `__syscall int k_msgq_put(struct k_msgq *msgq, void *data, k_timeout_t timeout);` and was definitively expecting data to be `const`.

Sadly it is not straightforward that `data` parameter can be constified in `z_impl_k_msgq_get` implementation:
https://github.com/zephyrproject-rtos/zephyr/blob/21756751997543fa6946d67415dce1f176f36d60/kernel/msg_q.c#L116
Indeed `data` is used as `src` in `memcpy` which support a `const`, but it is also used to store data ptr in `_current->base.swap_data`: https://github.com/zephyrproject-rtos/zephyr/blob/21756751997543fa6946d67415dce1f176f36d60/kernel/msg_q.c#L153

I'm not familiar with kernel swap_data usage and therefore I'm not able to see swap_data could be constified here: https://github.com/zephyrproject-rtos/zephyr/blob/21756751997543fa6946d67415dce1f176f36d60/include/kernel.h#L516

Does anyone known if this is possible?

PR #28452 (2020-09-23): doc: Clarify semantics of k_msgq_put
Amend Doxygen documentation for k_msgq_put to note that the message
content will not be modified as a result of the function call and
constify data parameter in function prototype.

Fixes #22301
commit f29a2d1c (2020-09-23): doc: Clarify semantics of k_msgq_put

Amend Doxygen documentation for k_msgq_put to note that the message
content will not be modified as a result of the function call and
constify data parameter in function prototype.

Fixes #22301
closed 2020-09-23 by MaureenHelm