zephyrproject-rtos/zephyr #22304: ARM Cortex-M STMF401RE: execution too slow

artem-hel, 2020-01-29:
**Describe the bug**
The execution of the Zephyr kernel code takes too much CPU cycles.
I am measuring CPU cycles gone while executing kernel code among different RTOS on the same nucleo-f401re board. Measurements are done with the same board. With other operating systems i can not observe same behavior. 

I have tried to set `CONFIG_SPEED_OPTIMIZATIONS=y` in prj.conf of the project, and with that i had achieved 20% less cycles, but the cycle count is still suspiciously big.

**To Reproduce**
1. Go to Folder containing ZephyrOS 
2. cd zephyrproject/zephyr/
2. west build -p auto -b nucleo_f401re tests/benchmarks/latency_measure/
3. west flash
4. See error

**Expected behavior**
I would expect significantly smaller numbers in the Benchmark output.

**Impact**
Can potentially couse a deadline miss which is the case with latency benchmark.

**Console output**
```
***** Booting Zephyr OS build zephyr-v2.0.0-1430-g9784f800a686 *****
|-----------------------------------------------------------------------------|
|                            Latency Benchmark                                |
|-----------------------------------------------------------------------------|
|  tcs = timer clock cycles: 1 tcs is 11 nsec                                 |
|-----------------------------------------------------------------------------|
| 1 - Measure time to switch from ISR back to interrupted thread              |
| switching time is 448 tcs = 5333 nsec                                       |
|-----------------------------------------------------------------------------|
| 2 - Measure time from ISR to executing a different thread (rescheduled)     |
| switch time is 498 tcs = 5928 nsec                                          |
|-----------------------------------------------------------------------------|
| 3 - Measure average time to signal a sema then test that sema               |
| Error: tick occurred                                                        |
| Average semaphore test time 75 tcs = 892 nsec                               |
|-----------------------------------------------------------------------------|
| 4- Measure average time to lock a mutex then unlock that mutex              |
| Average time to lock the mutex 93 tcs = 1107 nsec                           |
| Average time to unlock the mutex 252080 tcs = 3000955 nsec                  |
|-----------------------------------------------------------------------------|
| 5 - Measure average context switch time between threads using (k_yield)     |
| Error: tick occurred                                                        |
|-----------------------------------------------------------------------------|
| 6 - Measure average context switch time between threads (coop)              |
| Error: tick occurred                                                        |
|-----------------------------------------------------------------------------|
===================================================================
PROJECT EXECUTION FAILED
```

**Environment**
Ubuntu 7.4.0-1ubuntu1~18.04.1
GNU Arm Embedded Toolchain

andyross, 2020-01-29:
Those cycle counts don't look crazy awful to me (except the mutex one, no idea what you're seeing there): 4-500 cycles for a context switch, sub-100 for semaphore and mutex operations.

What is your reference target for what you are expecting to see?  I'm wondering if this is maybe just a board configuration glitch.  A quick check at ST's page for this board seems to say that the "HSE clock" should be 24MHz, but it's listed as CONFIG_CLOCK_STM32_HSE_CLOCK=8000000 in the PLL configuration for the board.  That would indeed make the measured 3x lower than the expected 84 MHz (see CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC).  Would a 3x shortfall in the nanosecond measurements (NOT the cycle counts, which come from SysTick and reflect CPU cycles) explain what you're seeing?

If so, do other timer-related tests seem to be going too slow also?

artem-hel, 2020-01-29:
@andyross, thank you for the quick reply!

I have also done context switching time measurements on the same **nucleo-f401re** board with 3 RTOS. Results are:

- ZephyrOS **1220** cycles
- FreeRTOS **409** cycles
- RIOTOS **341** cycles

To show, how the measurement were done, here is example from FreeRTOS realisation:
```
void ConSwitch1() {
	const TickType_t delay_ticks = pdMS_TO_TICKS(CON_SW_1_DELAY);

	while (1) {

		// ConSwitch2 only need during measurement
		vTaskResume(ConSwitch2Handle);

		start_measurement(M_DWT);
		taskYIELD();
		finish_measurement(M_DWT);
	
		vTaskSuspend(ConSwitch2Handle);

		vTaskDelay(delay_ticks);
	}
}

void ConSwitch2() {

	while (1) {
		taskYIELD();
	}
}
```

in `start_measurement(M_DWT)` and `finish_measurement(M_DWT)` i only read out `DWT->CYCCNT` register value and find out the difference. 

After that i have checked if it could be something wrong with my application so i just chose the default one from ZephyrOS tests folder to be sure. On the benchmark results there are also Errors because of tick interrupt. Does is not seem strange?

So i am curious if it is a normal behaviour.

andyross, 2020-01-29:
Can you post the code you used to measure the 1220 cycle context switch time on Zephyr and (especially!) the build/zephyr/.config file showing the kernel configuration.  Unfortunately I don't have an STM board, though for tests like this other Cortex-M systick boards would be expected to be similar I'd expect.

Regarding the benchmark output: yes, the "tick occurred" errors look weird, as does the mutex unlock result.   This benchmark doesn't get a lot of use, the CI system doesn't flag these errors as failures apparently (or they don't occur elsewhere, potentially) and it's possible something is bitrotten.

artem-hel, 2020-01-29:
Of course, here:

**main.c**
```
#include "stm32f401xe.h"
#include <zephyr.h>
#include <device.h>
#include <drivers/gpio.h>
#include <sys/printk.h>
#include <sys/__assert.h>
#include <string.h>

//time to sleep
#define ONE_SECOND		(MSEC_PER_SEC)

/* benchmark app header */
#include "../benchmark/Inc/application.h"

/* size of stack area used by each thread */
#define STACKSIZE 1024

/* scheduling priority used by each thread */
#define PRIORITY 7 

void blink1(void);
void blink2(void);
volatile int k = 0;

K_THREAD_DEFINE(blink1_id, STACKSIZE, blink1, NULL, NULL, NULL,
		PRIORITY, 0, K_NO_WAIT);
K_THREAD_DEFINE(blink2_id, STACKSIZE, blink2, NULL, NULL, NULL,
		PRIORITY, 0, K_NO_WAIT);

void blink1(void)
{

	CoreDebug->DEMCR |= CoreDebug_DEMCR_TRCENA_Msk;
	DWT->CYCCNT = 0;
	DWT->CTRL |= DWT_CTRL_CYCCNTENA_Msk;

	

	while(1){

	k_thread_resume(blink2_id);
	
	start_measurement(M_DWT);

	k_yield();
	
	finish_measurement(M_DWT);

	k_thread_suspend(blink2_id);

	print_meas_result(M_DWT);

	k_sleep(ONE_SECOND);

	}
}

void blink2(void)
{  
	
	while(1){

	k_yield();
	}
}

[... 794 lines cut ...]

}

...

void finish_measurement(int meas) {

	switch (meas) {

...

	case M_DWT:
...
		#ifdef ZEPHYR
		value.CYCLE_DIFF = DWT->CYCCNT - value.DWT_CYCLE_COUNTER;  // Time measurement with DWT Cycle Counter
		#endif // ZEPHYR
		break;
...

	default:
		break;
	}

}
```

andyross, 2020-01-29:
Nothing looks immediately wrong.  I do note that this platform builds with CONFIG_ARM_MPU=y despite the fact that neither userspace nor stack guard are enabled, which is incurring overhead that I'm guessing you're not seeing on the other platforms.  You could try turning that off to see if that  helps a bit.

Beyond that, it would be helpful to make this more portable (as mentioned, I don't have ST hardware locally) by using k_cycle_get_32() instead of your direct DWT reads.  On your board that is based on SysTick and should have the same precision modulo a little more overhead.  I'll try some of my other boards and see if the cycle counts I'm seeing there match yours.

But broadly this seems to be getting out of my area of expertise and into the depths of the ARM layer.  @erwango might be a better resource.

artem-hel, 2020-01-29:
Thank you for the participation, quick replys and a good hint!

After adding the lines: 
```
CONFIG_SPEED_OPTIMIZATIONS=y
CONFIG_ARM_MPU=n
```
to `prj.conf` in app directory i am now measuring **875** cycles instead of **1220** for context switch.
This is a good step in the right direction.

I hope, @erwango can have a look on this issue.

andrewboie, 2020-01-29:
I think there are some opportunities for optimization of context switching when the MPU is enabled, it really should just be saving/restoring the MPU register contents instead of doing all the calculations it is doing right now. cc @ioannisg 

875 still doesn't seem great compared to the other OSes you cited, so I think this is still worth investigating

artem-hel, 2020-01-29:
Actually after i had found this presentation from NXP: [BENCHMARK-ZEPHYR-OS-PDF.pdf](https://www.nxp.com/docs/en/training-reference-material/BENCHMARK-ZEPHYR-OS-PDF.pdf)
i have expected, that Zephyr will be the fastest RTOS among others.

Sure, m7 from presentation cannot be compared to m4 i am using in nucleo-f401re, but still. 
They get there **47** cycles for context switch using k_yield() on m7.

LMESTM, 2020-01-30:
My 2 cents: 
there are config options in tests/benchmarks/sys_kernel/prj.conf that may be useful to look at.
From quick tests I've done, maybe the CONFIG_FORCE_NO_ASSERT=y may improve performances.

artem-hel, 2020-01-30:
@LMESTM, thank you for a hint! 
Now after `CONFIG_FORCE_NO_ASSERT=y` adding in the prj.conf i measure **591** cycles instead of **875**.

erwango, 2020-01-30:
@artem-hel, I'd like to help but there is not much STM32 specific code that you're running in this benchmark. I hope `arch/arm` maintainer can help in this case.

artem-hel, 2020-01-30:
@erwango, thank you for reply.

It is also true that i do not involve peripherals specific for STM32 MCU's. 
The reason for that is simple:  i am trying to eliminate everithyng but the kernel handling of basic RTOS functionailty.

andyross, 2020-01-30:
Hm... assertions were supposed to have been historically disabled in the benchmarks directory.  That used to be true, anyway.  Those tools really need some attention.

ioannisg, 2020-01-30:
@artem-hel @andrewboie @andyross @erwango 
So I  ran a modified hello-world with the provided code snippet, and got the following results.I used the nrf5340 (Cortex-M33), which is similar to Cortex-M4.

  - default configuration: **871** cycles
  - disable all MPU re-programming at SWAP but keep the option enabled: **458-460** cycles
  - disable MPU: **454** cycles 
     - use FORCE_NO_ASSERT=y: **454** cycles
     - use SPEED_OPTIMIZATIONS=y: **390** cycles

ioannisg, 2020-01-30:
I am a bit puzzled with the **4-cycle** difference between ARM_MPU=n and when I simply disable all code under ARM_MPU in swap_helper.c. Also, I saw that this is not deterministic.

@andyross you assume correct that we have no USERSPACE or STACK GUARD, so the context-swich should not be affected by the MPU; the problem is that currently we clear the MPU at swap, anyway, so we pay the cycles penalty. Obviously, this could be fixed - I'll leave that for the 2.2 stabilization period.

ioannisg, 2020-01-30:
Btw, @artem-hel @andyross  I looked at the presentation of NXP and this magic 47 cycles of swap: I assume they might refer only to the execution of the pendSV handler. k_yield() involves all scheduling overhead of the kernel, as well.

@andrewboie what exactly is the bug here?

andrewboie, 2020-01-30:
I'm adding a bug tag here, I agree with others that some attention is needed.

@ioannisg what I'm thinking for the MPU is to defer the calculations of the MPU register contents to when the memory domains are programmed instead of at context switch. I made a similar change in x86 last summer. There's an enhancement ticket open for this already #15135

andrewboie, 2020-01-30:
> @andrewboie what exactly is the bug here?

@ioannisg the bug is the configuration of our defaults which is causing the reporter to have to force assertions off when running benchmarks. Someone needs to look into this more deeply.

The context switch speed wrt to MPU is an enhancement, already tracked.

ioannisg, 2020-01-30:
@artem-hel @andyross I used directly DWT inside swap to measure the cycles-duration of the swap function only, using the same setup:
- 260 cycles with default configuration
- **60 cycles** with MPU=n, which is very close to the value reported in the NXP draft.

ioannisg, 2020-01-30:
@andrewboie ok, then it's a test bug, I suppose. Maybe we need to re-describe the ticket.

As for the enhancement of the MPU, that's in my TODO list for next release.

ioannisg, 2020-01-31:
@artem-hel could you check whether #22357 makes the execution faster, at least for the given benchmark test you are referring to?

artem-hel, 2020-01-31:
@ioannisg,  thank you for replies.
i have measured again with the corrections from #22357 and could not observe any difference so far. i am still measuring **591** cycles for the context switch.
As before in my prj.conf i set:

```
CONFIG_SPEED_OPTIMIZATIONS=y
CONFIG_FORCE_NO_ASSERT=y
CONFIG_ARM_MPU=n
```

_answering earlier replies:_ 
> I used the nrf5340 (Cortex-M33), which is similar to Cortex-M4.

may i conclude, that your results on Cortex-M33 are better because of the improved core compared tom M4 and we actually get same result overall?

ioannisg, 2020-01-31:
>> may i conclude, that your results on Cortex-M33 are better because of the improved core compared to M4 and we actually get same result overall?

Going to repeat the tests on nRF Cortex-M4 and report back.

ioannisg, 2020-01-31:
@artem-hel nRF52840 (Cortex-M4) gives me
- 591 cycles for the above case, so this is exactly the number you 've got :)
- 70 cycles for the code swap() function, slightly more than CM-33.

To clarify: #22357 gives an improvement for the use-case you have MPU enabled (CONFIG_ARM_MPU=y). For disabled MPU this PR does no help. But I thought you might run the benchmark test which might have MPU enabled, so this would definitely improve the numbers.

@andyross @andrewboie what brings us from 70 cycles to 591 is whatever else is needed for swap (i.e. outside the ARM swap function), so we should probably look into the kernel implementation, what do you think?

artem-hel, 2020-01-31:
@ioannisg thank you and all participants for quick replies and engagement.

Now, after i know, that all possible adjustments are done and i could not do anything better, my question is answered. 
And so i can close the issue.

Wish to all of you nice weekend!

closed 2020-01-31 by artem-hel