zephyrproject-rtos/zephyr #22307: net: ip: net_pkt_pull(): packet corruption when using CONFIG_NET_BUF_DATA_SIZE larger than 256

a8961713, 2020-01-29:
**Describe the bug**
`CONFIG_NET_BUF_DATA_SIZE` has a default of 128.
When using `CONFIG_NET_BUF_FIXED_DATA_SIZE=y` and `CONFIG_NET_BUF_DATA_SIZE` > 256 (e.g. 512), `net_pkt_pull()` corrupts the packet, as it calculates the lengths using `u8_t`.

This can be reproduced when using ipv6 with packet fragmentation. The packet reassembling code uses `net_pkt_pull()` on received packets.
You can send large ipv6 pings (e.g. `ping6 -s 2000 192.0.2.2`) to see the corrupted data

**Impact**
This is a showstopper when working with fragmented IPV6 packets.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Version: current master

**Additional context**

The solution is to change the type of `left, rem` from `u8_t` to `size_t`:
```
diff --git a/subsys/net/ip/net_pkt.c b/subsys/net/ip/net_pkt.c
index dfd2596c8d..8ceb13ffea 100644
--- a/subsys/net/ip/net_pkt.c
+++ b/subsys/net/ip/net_pkt.c
@@ -1848,7 +1848,7 @@ int net_pkt_pull(struct net_pkt *pkt, size_t length)
        net_pkt_cursor_backup(pkt, &backup);
 
        while (length) {
-               u8_t left, rem;
+               size_t left, rem;
 
                pkt_cursor_advance(pkt, false);
```

tbursztyka, 2020-01-29:
Could you send a PR? 

When you do have a solution already, no need to open an issue first: you could describe what the commit is fixing in its message

a8961713, 2020-01-29:
OK, first thing tomorrow...

PR #22335 (2020-02-03): net: ip: net_pkt: Fix net_pkt_pull data buffer corruption
Fixes #22307
closed 2020-02-03 by jukkar