zephyrproject-rtos/zephyr #22316: samples/philosophers coop_only scenario times out on nsim_sem and nsim_em

andrewboie, 2020-01-29:
**Describe the bug**
This sample gets stuck and times out on both nsim_sem and nsim_em, causing a sanitycheck failure.

**To Reproduce**
Build samples/philosophers using the coop_only configuration:

```
  sample.kernel.philosopher.coop_only:
    extra_configs:
      - CONFIG_NUM_PREEMPT_PRIORITIES=0
      - CONFIG_NUM_COOP_PRIORITIES=6
```

Or just run `sanitycheck -p nsim_em -p nsim_sem -T samples/philosophers`

**Expected behavior**
Passing test. I used nSIM free 2017.09

vonhust, 2020-01-31:
@andrewboie  after configing CONFIG_NUM_PREEMPT_PRIOITIES=0, CONFIG_NUM_COOP_PRIORITIES=6,  i found the philosopher will fail in 
	Z_ASSERT_VALID_PRIO(priority, entry) of  arch_new_thread.  It's not only in arc , but aslo found in qemu_x86. Here is the log of qemu_x86

```
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 7
ASSERTION FAIL [((((priority)) == -1 && z_is_idle_thread_entry((entry))) || (((2
	invalid priority (0); allowed range: -2 to -8
```

I check the code, for coop_only case,  it seems the valid priorities should be < 0,  but CONFIG_MAIN_THREAD_PRIORITY is 0.

andrewboie, 2020-01-31:
Why is this test scenario passing on other boards that aren't ARC?

andrewboie, 2020-01-31:
cc @andyross

andrewboie, 2020-02-06:
> @andrewboie after configing CONFIG_NUM_PREEMPT_PRIOITIES=0, CONFIG_NUM_COOP_PRIORITIES=6, i found the philosopher will fail in
> Z_ASSERT_VALID_PRIO(priority, entry) of arch_new_thread. It's not only [...]

@vonhust I cannot reproduce your result.

For example if I make this change to prj.conf and run on qemu_x86 it works perfectly:

```
diff --git a/samples/philosophers/prj.conf b/samples/philosophers/prj.conf
index 667391194c..87aaa5de91 100644
--- a/samples/philosophers/prj.conf
+++ b/samples/philosophers/prj.conf
@@ -1,7 +1,9 @@
 CONFIG_STDOUT_CONSOLE=n
 CONFIG_ASSERT=y
 CONFIG_ASSERT_LEVEL=2
-CONFIG_NUM_COOP_PRIORITIES=29
-CONFIG_NUM_PREEMPT_PRIORITIES=40
 CONFIG_SCHED_SCALABLE=y
 CONFIG_MP_NUM_CPUS=1
+
+CONFIG_NUM_PREEMPT_PRIORITIES=0
+CONFIG_NUM_COOP_PRIORITIES=6
+
```

It hangs on ARC.
I really do think this is an ARC-specific issue unless you have more detailed steps to show this affects other arches.

abrodkin, 2020-02-13:
@andrewboie please upgrade your Free nSIM to the most recent version v2019.09
(available after click-through registration here https://www.synopsys.com/cgi-bin/dwarcnsim/req1.cgi).

PR #22778 (2020-02-13): arch: arc: fix the bug that idle thread blocks other threads
* for COOP_SCHED case, i.e., PREEMPT_ENABLED is not enabled, the
  idle thread will block other threads which is not correct.

* remove the check of PREEMPT_ENABLED in the epilogue of irq and
  exception handling. Let the scheduler (should_preempt()) decide
  whether the thread should be preempted.

Fixes #22316
closed 2020-02-13 by andrewboie