zephyrproject-rtos/zephyr #22322: Clang linking error

Akirakato1, 2020-01-29:
Followed https://github.com/zephyrproject-rtos/zephyr/pull/14077 way of changing tollchain:
export ZEPHYR_TOOLCHAIN_VARIANT=llvm
I have llvm-8-dev, clang, clang-8 installed through apt. 
for ZEPHYR_TOOLCHAIN_VARIANT=zephyr, sample/basic/blinky can build and make fine. Once swapped to llvm, the following error happens at build: 
```
-- The C compiler identification is Clang 6.0.0
-- The CXX compiler identification is Clang 6.0.0
-- The ASM compiler identification is Clang
-- Found assembler: /usr/bin/clang
-- Cache files will be written to: /root/.cache/zephyr
CMake Error at ../../../cmake/extensions.cmake:1457 (message):
  Assertion failed: The toolchain is unable to build a dummy C file.  See
  CMakeError.log.
Call Stack (most recent call first):
  ../../../CMakeLists.txt:42 (assert)

-- Configuring incomplete, errors occurred!
See also "/home/zephyrproject/zephyr/samples/basic/blinky/test/CMakeFiles/CMakeOutput.log".
See also "/home/zephyrproject/zephyr/samples/basic/blinky/test/CMakeFiles/CMakeError.log".
```

I can force this issue to not appear and build by feeding -D CMAKE_TRY_COMPILE_TARGET_TYPE=STATIC_LIBRARY when building. 

However, when I try to make, the following errors happen: 
```
In file included from /home/zephyrproject/modules/hal/nordic/nrfx/mdk/nrf.h:164:
/home/zephyrproject/modules/hal/nordic/nrfx/mdk/compiler_abstraction.h:154:16: warning:
      variable 'sp' is uninitialized when used here [-Wuninitialized]
        return sp;
               ^~
/home/zephyrproject/modules/hal/nordic/nrfx/mdk/compiler_abstraction.h:153:29: note:
      initialize the variable 'sp' to silence this warning
        register unsigned sp __ASM("sp");
                            ^
                             = 0
1 warning generated.
```
and at the very end of linking:
```
clang: warning: argument unused during compilation: '--specs=nosys.specs' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-nopie' [-Wunused-command-line-argument]
clang: error: unable to execute command: Executable "ld.lld" doesn't exist!
clang: error: ld.lld command failed with exit code 1 (use -v to see invocation)
zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:100: recipe for target 'zephyr/zephyr_prebuilt.elf' failed
make[2]: *** [zephyr/zephyr_prebuilt.elf] Error 1
CMakeFiles/Makefile2:2133: recipe for target 'zephyr/CMakeFiles/zephyr_prebuilt.dir/all' failed
make[1]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

Not sure what's wrong with it.

Akirakato1, 2020-01-31:
I swapped the symlink in usr/bin for clang to clang-8 and build can finish without CMAKE_TRY_COMPILE_TARGET_TYPE=STATIC_LIBRARY. Checking error log, the ld.lld error still persists but clang-8 somehow overcame that. Anyways, "make" errors are the same.

Akirakato1, 2020-01-31:
I just found https://github.com/galak/zephyr/tree/llvm but not sure how to use. My directory is set up as: [picture of direcory](https://gyazo.com/120dbca5f0af4199d390a3e69d182008). So first tried just cloning it to desktop, exporting the paths, and try building but "error cannot find file "nrf5x.h" at make. So I tried replacing this zephyr repo with the one inside ncs as "nrf5x.h" exists in module folder inside ncs (not inside folder zephyr). That threw west error at build. I then tried using the old zephyr folder but copying over changes you made by tracing your git commit but at every step it threw missing declaration or file so gave up on this thought. I used west to get my original zephyr going but have to git clone your llvm-arm fix?

Akirakato1, 2020-01-31:
@galak

galak, 2020-02-06:
@Akirakato1 are you able to join slack?  If so the Zephyr SDK channel is an easier and quicker way for us to converse.

Akirakato1, 2020-02-06:
Unfortunately no, I do not have one of the whitelisted emails.

galak, 2020-02-06:
> Unfortunately no, I do not have one of the whitelisted emails.

@carlescufi you know a bit more about the slack invites..

galak, 2020-02-06:
> However, when I try to make, the following errors happen:
>
> ```
> In file included from /home/zephyrproject/modules/hal/nordic/nrfx/mdk/nrf.h:164: [...]

I submitted a PR to the nrf hal to address this.  https://github.com/zephyrproject-rtos/hal_nordic/pull/22

carlescufi, 2020-02-06:
@Akirakato1 please use this link to join slack:
https://tinyurl.com/y5glwylp

closed 2020-07-31 by github-actions[bot]