zephyrproject-rtos/zephyr #22337: hal: silabs: Clarify best source for silabs hal files

chrta, 2020-01-30:
## Introduction

This issue should continue the discussion of https://github.com/zephyrproject-rtos/hal_silabs/pull/3 and collect all relevant information for this.

### Problem description

It is not 100% clear where to pull the files from to update https://github.com/zephyrproject-rtos/hal_silabs There are multiple sources that have some pros and cons.

Also the version information about the contained files needs clarification.

The current flow is described in https://github.com/zephyrproject-rtos/hal_silabs/blob/master/gecko/README

Things that should be solved in this issues are:

* [ ] what are the possible/preferred sources for CMSIS device files and the emlib?
* [ ] how can we ensure version compatibility between them?
* [ ] what version information should we use to identify the version of the files in this hal (2.4 vs 5.6.1.0)?
* [ ] update README accordingly
* [ ] update this hal to the most recent version using the documented way

### Proposed change

To be discussed in this issue.

chrta, 2020-01-30:
@conq Can you comment here or get someone from SiLabs involved to help answering the above questions?

carlescufi, 2020-01-30:
CC @rettichschnidi

chrta, 2020-01-30:
Information regarding Simplicity Studio and its SDKs:
For my installation this is shown:
[image]

There is a "Gecko SDK Suite" version 2.7.1. The "32-Bit MCU SDK" has version 5.9.1.0. The "Flex SDK" has Version 2.7.1.0. All of these files are stored in the folder .../gecko_sdk_suite/v2.7

I never installed "Gecko SDK Suite" directly, so i assume that both "32-Bit MCU SDK" and "Flex SDK" depend on this package.

So i think for version information the version of the "Gecko SDK Suite" should be used instead of the version of the "32-Bit MCU SDK".

stevew817, 2020-03-25:
I'll try to answer here considering @conq is on leave for a while.

- We currently only have one authoritative source for our SDKs, and that is downloading through Simplicity Studio. We are discussing deployment in other ways, too, but that will not be resolved soon.
- Our CMSIS packs only provide CMSIS device headers, reference manuals / datasheets and debug support (SVD files and flashloaders). SDK software such as emlib, emdrv or RAIL is not distributed through the pack mechanism.
- The same content as contained in the CMSIS pack can also be found within the SDK download, and is part of the 'Gecko SDK Suite' base package. This is the foundation for all SDKs, so SDKs like 'Flex SDK' and '32-Bit MCU SDK' will automatically pull in the correct 'Gecko SDK Suite' version if not installed yet.
- Version compatibility is guaranteed within an SDK Suite version (the version number in the path). Currently, a new SDK version is scheduled on a bi-yearly schedule, though this is of course subject to change. Updates within the same SDK Suite version are regarded as patches and will contain no breaking changes.

mnkp, 2020-03-26:
Thanks for the clarification. The current version of SiLabs hal Zephyr is using (CMSIS device headers + emlib) is quite outdated. I was planning to bring it up to date and about to ask if it's already available outside of Simplicity Studio. 

I will use Simplicity Studio SDK for now. However, if the files will be published as CMSIS packs or in some other way it would be great to hear it.

closed 2023-04-13 by nashif