zephyrproject-rtos/zephyr #22340: Security problem with settings and littlefs

Laczen, 2020-01-30:
## Introduction

Settings is used to store security related information for bluetooth (e.g. bluetooth mesh keys, ...), this information is stored unencrypted.
Settings can use multiple backends to store this information: nvs, fcb, littlefs.
MCUmgr allows image management, device management and also file management. The file management can be used to copy files from a device. This file management is at the moment limited to littlefs.

### Problem description

When settings is used in combination with littlefs all security related information can be extracted from the device using MCUmgr and this could be used e.g in bt-mesh to get the device key, network key, app keys from the device.

### Proposed change

There are multiple ways to remove this security issue:

a. Do not allow littlefs as a backend for security related settings.
b. Encrypt the settings file so it cannot be used.
c. Replace settings with another mechanism to store security related information.
d. Remove file management from MCUmgr.
...

nashif, 2020-03-25:
@ceolin @d3zd3z  Can you please take a look at this and triage accordingly..

nixward, 2020-03-25:
Hi @Laczen,
For mcumgr file management over BLE see my PR that added a default authenticated requirement for the SMP/mcumgr characteristic: https://github.com/zephyrproject-rtos/zephyr/pull/20557

If the attacker has physical access to the device this possibly won't help as the attacker may have the access requirements for authentication but it's better than allowing mcumgr access to anyone who can make a BLE connection to the device.

The FCB backend for settings has the same physical access security issue.  The FCB partition contents could be read via JTAG and the keys extracted.  This assumes the FCB partition is internal to the SoC, but with other methods FCB settings in an external flash IC could be read as well.

BLE mesh allows the keys to be updated regularly to increase security against a rubbish bin attack (if the network security requires it) but (AFAIK) it wouldn't protect the network if the keys were currently in use.

The nRF5340 and other SoC's have hardware secure key storage but I'm not familiar it or how Zephyr is currently using it or planning to.  It's also up to the application developer to ensure they choose hardware with secure key storage and ensure it's used if the application requires this level of security.

Laczen, 2020-03-25:
@nixward, thanks for the reply.

The addition of a authenticated requirement for the ble smp/mcumgr characteristic is a good improvement. But is it enough ? When settings is used to store security related information it should IMO be impossible to read the settings data. It is correct that it should also be enforced that security related settings are not stored on external devices.

I am mainly worried that users are unaware of these problems as they are not documented.

JTAG access is a separate item, as soon as this is available the solution is unsecure IMO.

d3zd3z, 2020-03-25:
Some possible ideas for a fix: 1. Prevent the access. 2. Encrypt. This can be difficult without HW support for HW-unique keys. Otherwise, it is usually pretty trivial for an attacker to get the key being used for the encryption. 3. Document the configuration combinations that are not allowed (possibly checking them in the security config lint).

This probably warrants being disclosed as a CVE. I'll create a ZEPSEC issues to track that.

d3zd3z, 2020-03-25:
In addition to the above, another possible fix would be to disallow certain directories/files from being accessed through the file management of MCUmgr.

ceolin, 2020-06-08:
@Laczen is there a real use case where file management is needed ? This insecure in many ways, even if we come up with some ACL limiting access to specific directories (that btw, has to be done in MCU manager) there will still exists security holes, like starving the file system. From my understanding the proper way to solve it would be authenticate access, but that introduce other level of complexity. For the keys, is exactly what @d3zd3z said, without hw support to securely store the key used to encrypt it is useless.

Laczen, 2020-06-08:
@ceolin, IMO the best option is not to allow file access over mcumgr.

PR #26083 (2020-06-12): mcumgr fs management security information
Add security information about enabling fs management for mcumgr.

Closes #22340
closed 2020-06-12 by carlescufi