zephyrproject-rtos/zephyr #22356: An application hook for early init

susundberg, 2020-01-30:
**Is your feature request related to a problem? Please describe.**
STM32 platform has system-bootloader - that can be jumped into [1]. The jump must either be done during early init (for example before interupt handlers are placed, or system clock modified SIC) or the tear down must be handled with care (for example unregistere / mask all interrupt handlers). 

I find it easier to do right after init - as i find it simple with no need for careful tear down. 

Example session:
1. During normal application run write to backup [2] register specific value "jump to bootloader" 
2. Call system reset.
3. During early init, check value of backup-register. If value is "jump to bootloader" reset the value and jump to bootloader.
4. STM32-System bootloader can now be used (for example UART + CAN)

**Describe the solution you'd like**
I would like to have application hook that overrides default `__weak__` function placed in `kernel/init.c `for example in `z_cstart()` after `gcov_static_init` or similar. Saying for example as i am not sure about architecture / boot procedure.

Say naming could be `k_sys_early_init` or similar.

**Describe alternatives you've considered**

* Some `#define` trick - but i find weak cleaner.
* Misusing `gcov_static_init` - but that causes trouble with initialization and is not intuitive

**Additional context**
I suggested `__weak__` overide as `k_sys_fatal_error_handler` is done in similar fashion.

I know the stm32 system bootloader is platform specific feature - and there exists also more global solution of using proper flashed bootloader - but for example stm32_min_dev board with 64k flash is pretty tight with shell and logging -- my minimal application takes around 60% of flash.

[1] https://community.st.com/s/question/0D50X00009Xkg9mSAB/jump-to-bootloader-area-from-user-application-on-stm32f0
[2] https://os.mbed.com/users/gregeric/notebook/using-stm32-rtc-backup-registers/

andrewboie, 2020-01-30:
Why doesn't implementing your own function hooked into the existing kernel init framework with SYS_INIT() work for you?

susundberg, 2020-01-31:
My bad, i had not found that from the documentation.

If somebody ever finds this with search the stm32 bootloader entry works fine with:
`SYS_INIT( stm32_bootloader_check, PRE_KERNEL_1, 0 );`. At least on my tests the priority `CONFIG_KERNEL_INIT_PRIORITY_DEFAULT` did not work.

Closing the ticket. Thanks for your time.

closed 2020-01-31 by susundberg

susundberg, 2020-05-18:
For the record, if anybody else stumbles upon this (i got slack asked so i document it here):

Check: [boot.c](https://github.com/susundberg/zephyr-robot-supa2019/blob/f5d29812a53d1e41149dff5043dc086e1e15d716/src/utils/boot.c) and [main.c](https://github.com/susundberg/zephyr-robot-supa2019/blob/f5d29812a53d1e41149dff5043dc086e1e15d716/src/main.c#L272) from my repository for working example.