zephyrproject-rtos/zephyr #22366: Bug in sockets.c (subsys\net\lib\sockets)

bran795, 2020-01-31:
**Describe the bug**
A clear and concise description of what the bug is.
When zsock_accept_ctx is called on a non-clocking socket, after a few times it returns ENFILE - no more file descriptors available
What have you tried to diagnose or workaround this issue?
Call 		
z_free_fd(fd);
Before returning EAGAIN (line 16 of the function accept)

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Accept should return EAGAIN when there is no pending connection on the listening socket

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
Showstopper. This bug means non-blocking sockets cannot be used for listening and accepting on the server side.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.
N/A

**Environment (please complete the following information):**
 - OS:  Windows
 - Toolchain arm mbed
 - Commit SHA or Version used: top as of Jan 30 2020

**Additional context**
Add any other context about the problem here.
code snippet:

@@ -421,6 +421,7 @@ int zsock_accept_ctx(struct net_context *parent, struct sockaddr *addr,
	ctx = k_fifo_get(&parent->accept_q, timeout);
	if (ctx == NULL) {
		errno = EAGAIN;
+		z_free_fd(fd);
		return -1;
	}

jhedberg, 2020-02-04:
@bran795 are you planning to send a pull request, since you seem to have a fix proposal in the description?

bran795, 2020-02-04:
@jhedberg Hi, newbie here
I haven't checked the procedure for sending a pull request
Can anyone send a pull request?

bran795, 2020-02-06:
@jhedberg
I would appreciate a little help with the pull request
Is there a way to edit the pull request comment after it has been created?
Thanks

pfalcon, 2020-02-06:
@bran795: "Comment", not (well, it won't help). "Commit message", yes. Use `git rebase -i master` in your branch. Read instructions carefully, and feel free to google it up for more details (Zephyr docs may also include suggestions for development process).

bran795, 2020-02-06:
@pfalcon right
what about the Identity/Emails issues? I added an email address first.last@company to my github account but that doesn't seem to work either
thanks

jhedberg, 2020-02-06:
@bran795 you need to configure your local git correctly, which is normally done through `$HOME/.gitconfig`. E.g. in mine I have:
```
[user]
	name = Johan Hedberg
	email = johan.hedberg@intel.com
```

Since you already created the commit you'll need to amend its author explicitly. You should be able to do that with `git commit --amend --author="Your Name <your@email>` or with `git commit --amend --reset-author` after you've fixed your git configuration.

pfalcon, 2020-02-06:
@bran795: Name/email in commit's "Author:" field should match those in "Signed-off-by:" as present in the commit message. In #22366 I see (`git log`):

~~~
Author: bran795 <48238460+bran795@users.noreply.github.com>
~~~

which suggests that you submitted that PR from Github's web UI. That won't work here, you would need to use command-line git client, I'm afraid.

bran795, 2020-02-06:
@jhedberg 
@pfalcon 
Thanks, I'll install git and try again

jhedberg, 2020-02-06:
> you submitted that PR from Github's web UI

I had no idea that GitHub had such a feature. I wonder if it can be disabled per-project, since it doesn't seem to be able to produce what's required for Zephyr.

bran795, 2020-02-06:
Getting closer...

Gitlint issues
1: UC2 Signed-off-by: must have a full name

Identity/Emails issues
1a868e8: author email (Inbar Anson Bratspiess inbar.anson.bratspiess@330plus.net) needs to match one of the signed-off-by entries.

Please help 

Thanks

PR #22579 (2020-02-10): net: socket: Release fd on failure in accept()
The zsock_accept_ctx() calls z_reserve_fd() on entry but fails
to call z_free_fd() on failure. This will leak the allocated
socket descriptor.

Fixes #22366
commit 827f6f17 (2020-02-10): net: socket: Release fd on failure in accept()

The zsock_accept_ctx() calls z_reserve_fd() on entry but fails
to call z_free_fd() on failure. This will leak the allocated
socket descriptor.

Fixes #22366
closed 2020-02-10 by jukkar