zephyrproject-rtos/zephyr #22379: stm32: can_send gets stuck (ignoring timeout) if bus not present

martinjaeger, 2020-01-31:
If the transceiver is not wired to the bus, `can_stm32_send` gets stuck after 3 calls (= length of the internal TX queue of the used STM32F072) in the following line:
https://github.com/zephyrproject-rtos/zephyr/blob/ac12066800ac0d7c2e353462b830f96e8e97fe1b/drivers/can/can_stm32.c#L630

Changing the  `k_sem_take` timeout from `K_FOREVER` to `timeout` (= function argument) solved the problem. But I'm not sure if this is the ideal solution, @alexanderwachter .

I'm currently using the CAN interface in a very simple way without any callbacks:
`can_send(can_dev, &msg, K_MSEC(10), NULL, NULL);`

alexanderwachter, 2020-01-31:
We are already aware of this situation.
Tho problem is that the frame is in the mailbox already and we cant remove it from there.
#20890 is the fix for it.

martinjaeger, 2020-01-31:
Great, thanks. And sorry that I didn't look carefully enough in the open PRs.

alexanderwachter, 2020-01-31:
> Great, thanks. And sorry that I didn't look carefully enough in the open PRs.

No problem. I always try to lable all CAN related PRs and Issues with "area: CAN"

henrikbrixandersen, 2022-01-18:
@martinjaeger I would say this is the expected behaviour. The timeout in `can_send()` is for acquiring a TX MB, not a transmission timeout (as such a transmission timeout would be rather difficult to honour on many CAN controllers, which tend to always retransmit in hardware). By default, the CAN controller will attempt to retransmit frames without ACK or when arbitration is lost, which would happen if no CAN transceiver/CAN bus is connected.

With the recent changes to the API documentation in #41845, hopefully this behaviour is more clear. The same PR also introduced "one-shot" mode for the STM32 CAN controller, which will disable the automatic retransmission in hardware and perhaps be closer to the behaviour, you expect?

martinjaeger, 2022-01-18:
I agree that it can be considered expected behaviour, so we can probably close the issue.

As far as I can see, the one-shot mode can only be configured via Devicetree. If that's the case, it is not useful for my application, as I want to use the same CAN interface either in "fire & forget" mode for low priority messages or with proper error checking for ISO-TP. But it's not a big deal, I created an empty callback so that `can_send` still returns if the mailbox is full because of transmission errors.

closed 2022-01-18 by henrikbrixandersen