zephyrproject-rtos/zephyr #22388: Build warnings in http_get on cc3220sf_launchxl

vanti, 2020-01-31:
**Describe the bug**
Build warnings are seen when building http_get sample for cc3220sf_launchxl in CI:

18:30:10  NET_CONFIG_MY_IPV4_GW (defined at subsys/net/lib/config/Kconfig:104) was assigned the value
18:30:10 '192.0.2.2' but got the value ''. Check these unsatisfied dependencies: NET_CONFIG_SETTINGS (=n).
18:30:10 See http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_NET_CONFIG_MY_IPV4_GW.html and/or
18:30:10 look up NET_CONFIG_MY_IPV4_GW in the menuconfig/guiconfig interface. The Application Development
18:30:10 Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
18:30:10 might be helpful too.
18:30:10 
18:30:10 
18:30:10 warning: NET_SOCKETS_POSIX_NAMES (defined at subsys/net/lib/sockets/Kconfig:13) was assigned the
18:30:10 value 'n' but got the value 'y'. See
18:30:10 http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_NET_SOCKETS_POSIX_NAMES.html and/or
18:30:10 look up NET_SOCKETS_POSIX_NAMES in the menuconfig/guiconfig interface. The Application Development
18:30:10 Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
18:30:10 might be helpful too.
18:30:10 
18:30:10 
18:30:10 warning: TLS_CREDENTIAL_FILENAMES (defined at subsys/net/lib/tls_credentials/Kconfig:17) was
18:30:10 assigned the value 'y' but got the value 'n'. Check these unsatisfied dependencies: TLS_CREDENTIALS
18:30:10 (=n). See
18:30:10 http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_TLS_CREDENTIAL_FILENAMES.html and/or
18:30:10 look up TLS_CREDENTIAL_FILENAMES in the menuconfig/guiconfig interface. The Application Development
18:30:10 Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
18:30:10 might be helpful too.
18:30:10 
18:30:10 
18:30:10 warning: NET_SOCKETS_POSIX_NAMES (defined at subsys/net/lib/sockets/Kconfig:13) has direct dependencies !POSIX_API && NET_SOCKETS && NETWORKING with value n, but is currently being y-selected by the following symbols:
18:30:10  - NET_SOCKETS_OFFLOAD (defined at subsys/net/lib/sockets/Kconfig:102), with value y, direct dependencies NET_SOCKETS && NETWORKING (value: y), and select condition NET_SOCKETS && NETWORKING (value: y)
18:30:10 
18:30:10 error: Aborting due to Kconfig warnings

For full log, see:
https://ci.linaro.org/job/zephyr-upstream/PLATFORM=cc3220sf_launchxl,ZEPHYR_TOOLCHAIN_VARIANT=zephyr,label=docker-xenial-amd64-13/4801/console

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr/samples/net/sockets/http_get
2. west build -b cc3220sf_launchxl -p
3. See warnings

**Expected behavior**
There should be no build warnings.

**Impact**
Breaks CI.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.11.0
 - Commit SHA or Version used: 9dc4eed622e642e2423e2c71526b95f2b1c1ab08

ulfalizer, 2020-01-31:
This one is an issue with the Kconfig files (more select shenanigans):

```
warning: NET_SOCKETS_POSIX_NAMES (defined at subsys/net/lib/sockets/Kconfig:13)
has direct dependencies !POSIX_API && NET_SOCKETS && NETWORKING with value n,
but is currently being y-selected by the following symbols:
 - NET_SOCKETS_OFFLOAD (defined at subsys/net/lib/sockets/Kconfig:102), with
   value y, direct dependencies NET_SOCKETS && NETWORKING (value: y), and
   select condition NET_SOCKETS && NETWORKING (value: y)
```

Will look into later.

vanti, 2020-01-31:
> This one is an issue with the Kconfig files (more select shenanigans):
>
> ```
> warning: NET_SOCKETS_POSIX_NAMES (defined at subsys/net/lib/sockets/Kconfig:13)
> has direct dependencies !POSIX_API && [...]

@ulfalizer I had proposed a change in #22385 to address this one, since that test doesn't really apply to cc32xx imho. If it is due to something else, keep me posted and I'll remove that commit. Thanks!

ulfalizer, 2020-01-31:
@vanti 
When there's selects with unsatisfied dependencies, it's always a problem within the Kconfig files, because it means dependencies aren't properly specified.

Best to think of Kconfig in terms of the menuconfig. It shouldn't be possible to go into the menuconfig and create weird configurations where symbols are enabled but not their dependencies.

A quick workaround is fine though.

PR #22385 (2020-02-05): samples: http_get: fix build warnings on cc32xx
This PR eliminates the Kconfig warnings seen in build by setting the
options to the appropriate values for cc32xx platforms. They were
causing CI failures.

I also modified sample.yaml to exclude build for cc32xx for subtest
sample.net.sockets.http_get.posix, which does not apply.

Fixes #22388
commit 801077f0 (2020-02-05): samples: http_get: fix build warnings on cc32xx

This PR eliminates the Kconfig warnings seen in build by setting the
options to the appropriate values for cc32xx platforms. They were
causing CI failures.

Fixes #22388
closed 2020-02-05 by jukkar