zephyrproject-rtos/zephyr #22411: AArch64 / Cortex-A port improvements / TODO

carlocaione, 2020-02-03:
https://github.com/zephyrproject-rtos/zephyr/pull/20263 introduced the basic support for AArch64 / Cortex-A.

This issue is used to track the TODO / WiP work.

**WiP**:
- MMU support: https://github.com/zephyrproject-rtos/zephyr/pull/20263#issuecomment-568964341 and https://github.com/zephyrproject-rtos/zephyr/issues/22410 (@Abhishek-brcm)
- Support for Xilinx ZynqMP APU (@carlocaione)

**TODO**:
- Merge shared code between Cortex-R / Cortex-A
- Support GICv3
- Rework interrupt handling (remove cascade IRQ controller). Reuse Cortex-R code from https://github.com/zephyrproject-rtos/zephyr/pull/19698
- Support SMP and switch to `arch_switch()`
- Do not update sp with each STP instruction (https://github.com/zephyrproject-rtos/zephyr/pull/20263#issuecomment-577606572)

carlocaione, 2020-02-03:
tagging interested / relevant people: @stephanosio @Abhishek-brcm @sandeepbrcm @timork @Mani-Sadhasivam @andrewboie @Jojjec1 

Feel free to add / modify the list.

carlocaione, 2020-12-01:
This is outdated. Closing for now.

closed 2020-12-01 by carlocaione

JohnCarterGonzalez, 2024-01-26:
Am I correct in assuming that because this issue is closed it means that Zephyr has Cortex A family support?

carlocaione, 2024-01-30:
> Am I correct in assuming that because this issue is closed it means that Zephyr has Cortex A family support?

yes