zephyrproject-rtos/zephyr #22420: onoff service: Extract transition functions to dedicated structure.

nordic-krch, 2020-02-03:
**Is your enhancement proposal related to a problem? Please describe.**
Currently transition functions (3 pointers) are part of `struct onoff_service` which must be stored in RAM. If onoff service is used for multiple instances then it is likely that transition functions are the same, given that `context` can be passed as an argument (#22419).

**Describe the solution you'd like**

It would be better to store pointers in flash and reuse it between onoff service instances. This will save RAM (4 bytes instead of 12 bytes). Transition function pointers should be moved to dedicated structure and onoff structure should keep a pointer.

It could optimized even further by splitting RAM part into control_block, allowing `struct onoff_service` to be declared in flash, too. Saving 12 bytes of RAM per instance.

nordic-krch, 2020-02-03:
@pabigot if you have no objections i'll prepare PR.

pabigot, 2020-02-03:
I do have objections.  It seems #22419 is required only because of this proposal, which provides a small savings in space for a special case where the functions are useful for multiple services.

While this could save a few bytes of RAM per service in this case, in the whole it seems to add complexity.  I believe the number of times there will be multiple service instances that can share a function table will be limited.  That belief may be wrong, but we need more thought before rushing into a refactoring, especially since it won't make 2.2 anyway.

nordic-krch, 2020-02-03:
> While this could save a few bytes of RAM per service in this case, in the whole it seems to add complexity.

What complexity?

> It seems #22419 is required only because of this proposal,

i see this as generic case when you have a multi-instance module which is using onoff service underneath and each instance needs to keep in RAM 3 the same, constant pointer.

pabigot, 2020-02-03:
I've reviewed #22419 with no objections in principle.  As the only constant parts of `struct onoff_service` are the function pointers I don't think further refactoring is likely to work.  (`flags` may look constant, but it isn't necessarily so.)

It would be reasonable to add the commit that pulls out the function pointers to a separate structure into #22419 as they are intertwined in purpose.

pabigot, 2020-02-04:
Following [discussion in #22419](https://github.com/zephyrproject-rtos/zephyr/pull/22419#issuecomment-581916399) please proceed with refactoring `struct onoff_service` so that the transition functions are stored in a separate structure that can be placed in RAM, while the rest remains as-is in RAM.  That should save RAM in all cases, and total memory whenever more than one service instance is used.

kartben, 2023-08-20:
Fixed with #23601

closed 2023-08-20 by kartben