zephyrproject-rtos/zephyr #22421: mbed TLS: Inconsistent Kconfig option names

mrfuchs, 2020-02-03:
While working on #22407 I noticed some naming inconsistencies in mbed TLS Kconfig options:

`MBEDTLS_CCM_C` was translated to `CONFIG_MBEDTLS_CIPHER_CCM_ENABLED`, but
`MBEDTLS_GCM_C` was translated to `CONFIG_MBEDTLS_CIPHER_MODE_GCM_ENABLED`, while
`MBEDTLS_CIPHER_MODE_CBC` was translated to `CONFIG_MBEDTLS_CIPHER_CBC_ENABLED`, but `MBEDTLS_CIPHER_MODE_XTS` was translated to `CONFIG_MBEDTLS_CIPHER_MODE_XTS_ENABLED`.

Also, some mbed TLS cipher mode options cannot be set by Kconfig, like `MBEDTLS_CIPHER_MODE_CFB`, `MBEDTLS_CIPHER_MODE_OFB` and `MBEDTLS_CIPHER_MODE_CTR`. That's probably the reason why the current version of the mbed TLS shim driver does not support AES-CTR mode:

https://github.com/zephyrproject-rtos/zephyr/blob/814654af80fefe21968717f32eefa34ec27aadd2/drivers/crypto/crypto_mtls_shim.c#L251-L254
https://github.com/zephyrproject-rtos/zephyr/blob/814654af80fefe21968717f32eefa34ec27aadd2/drivers/crypto/crypto_mtls_shim.c#L290-L291

rlubos, 2020-02-05:
I could either prefix each Kconfig with `MBEDTLS_CIPHER_MODE` or follow the mbedTLS convention (still, I'd prefer to keep the full `MBEDTLS_CIPHER` prefix). Will the following work for you?

| mbedTLS | Kconfig |
| --- | --- |
| `MBEDTLS_CCM_C` | `CONFIG_MBEDTLS_CIPHER_CCM_ENABLED` |
| `MBEDTLS_GCM_C ` | `CONFIG_MBEDTLS_CIPHER_GCM_ENABLED` |
| `MBEDTLS_CIPHER_MODE_CBC ` | `CONFIG_MBEDTLS_CIPHER_MODE_CBC_ENABLED` |
| `MBEDTLS_CIPHER_MODE_XTS ` | `CONFIG_MBEDTLS_CIPHER_MODE_XTS_ENABLED` |

As for the Kconfig mbedTLS options - everyone is free the extend the Kconfig option list per his/her requirements. The original file was created to cover the functionalities needed by the TLS sockets implementation and is gradually updated with new features. Feel free to contribute if you find something missing.

mrfuchs, 2020-02-05:
Looks good. I'm fine with the proposed naming fixes.
I agree that gradually adding required options makes sense.

PR #22503 (2020-05-08): modules: mbedtls: Update mbedTLS commit and apply fixes
Update mbedTLS commit along with the following fixes:

* Fix naming inconsistencies in some cipher modes, to match core mbedTLS
  configs
* Add Kconfig to enable CTR cipher mode

Fixes #22421
commit 2c277197 (2020-05-08): modules: mbedtls: Update mbedTLS commit and apply fixes

Update mbedTLS commit along with the following fixes:

* Fix naming inconsistencies in some cipher modes, to match core mbedTLS
  configs
* Add Kconfig to enable CTR cipher mode

Fixes #22421
closed 2020-05-08 by galak