zephyrproject-rtos/zephyr #22430: [Coverity CID :207985] Argument cannot be negative in subsys/net/lib/websocket/websocket.c

zephyrbot, 2020-02-03:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/a3e89e84a801d9bc048b0ee2177f0fb11d1a925a/subsys/net/lib/websocket/websocket.c#L789

Category: Memory - corruptions
Function: `websocket_recv_msg`
Component: Networking
CID: [207985](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=207985)

Details:

```
697     #else
698             ret = recv(ctx->real_sock, &ctx->tmp_buf[ctx->tmp_buf_pos],
699                    ctx->tmp_buf_len - ctx->tmp_buf_pos,
700                    timeout == K_NO_WAIT ? MSG_DONTWAIT : 0);
701     #endif /* CONFIG_NET_TEST */
702    
>>>     CID 207985:    (REVERSE_NEGATIVE)
>>>     You might be using variable "ret" before verifying that it is >= 0.
703             if (ret < 0) {
704                 return -errno;
705             }
706    
707             if (ret == 0) {
708                 /* Socket closed */
783             ret = input_len;
784     #else
785             ret = recv(ctx->real_sock, ctx->tmp_buf, ctx->tmp_buf_len,
786                    timeout == K_NO_WAIT ? MSG_DONTWAIT : 0);
787     #endif /* CONFIG_NET_TEST */
788    
>>>     CID 207985:    (REVERSE_NEGATIVE)
>>>     You might be using variable "ret" before verifying that it is >= 0.
789             if (ret < 0) {
790                 return -errno;
791             }
792    
793             if (ret == 0) {
794                 return 0;
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jukkar, 2020-02-12:
False positive, Coverity gets confused by various #ifdefs

closed 2020-02-12 by jukkar

jenmwms, 2020-02-13:
@jukkar Thanks for looking into this and offering rationale. It helps me learn!