zephyrproject-rtos/zephyr #22431: [Coverity CID :207984] Sizeof not portable in drivers/counter/counter_handlers.c

zephyrbot, 2020-02-03:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/a3e89e84a801d9bc048b0ee2177f0fb11d1a925a/drivers/counter/counter_handlers.c#L71

Category: Code maintainability issues
Function: `z_vrfy_counter_get_value`
Component: Drivers
CID: [207984](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=207984)

Details:

```
65     #include <syscalls/counter_ticks_to_us_mrsh.c>
66    
67     static inline int z_vrfy_counter_get_value(struct device *dev,
68                            u32_t *ticks)
69     {
70         Z_OOPS(Z_SYSCALL_DRIVER_COUNTER(dev, get_value));
>>>     CID 207984:  Code maintainability issues  (SIZEOF_MISMATCH)
>>>     Passing argument "ticks" of type "u32_t *" and argument "4U /* sizeof (ticks) */" to function "arch_buffer_validate" is suspicious. In this case, "sizeof (u32_t *)" is equal to "sizeof (u32_t)", but this is not a portable assumption.
71         Z_OOPS(Z_SYSCALL_MEMORY_WRITE(ticks, sizeof(ticks)));
72         return z_impl_counter_get_value((struct device *)dev, ticks);
73     }
74     #include <syscalls/counter_get_value_mrsh.c>
75    
76     static inline int z_vrfy_counter_set_channel_alarm(struct device *dev,
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

henrikbrixandersen, 2020-02-06:
I am not sure why this was flagged by coverity? Can somebody please explain?

nordic-krch, 2020-02-07:
also have no idea.

ceolin, 2020-03-23:
`sizeof(ticks)` will return the size of a pointer that can be different from the size of u32_t. In 64 bits architecture pointers are 8 bytes long and u32_t is 4 bytes.

ceolin, 2020-03-23:
Fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/23698

PR #23698 (2020-03-25): Coverity fixes for #22431 and #22643
General fixes for problem spotted by Coverity.

Fixes #22431 - z_vrfy_counter_get_value should check the size of memory pointed to
ticks and not the size of the pointer.

Fixes #22643 - gpio_pin_get() returns a negative value in case of error and
callbacks_configure was assigning this value to an unsigned variable.
commit 81d96d6e (2020-03-25): drivers: counter: Fix get_value verification handler

z_vrfy_counter_get_value should check the size of memory pointed to
ticks and not the size of the pointer.

Fixes: #22431
Coverity CID :207984
closed 2020-03-25 by ioannisg