zephyrproject-rtos/zephyr #22442: [Coverity CID :207965] Missing break in switch in drivers/i2c/i2c_ll_stm32_v1.c

zephyrbot, 2020-02-03:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/a3e89e84a801d9bc048b0ee2177f0fb11d1a925a/drivers/i2c/i2c_ll_stm32_v1.c

Category: Control flow issues
Function: `handle_rxne`
Component: Drivers
CID: [207965](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=207965)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

ydamigos, 2020-02-04:
I don't have access to coverity's report, but if it is referring to lines 293-295, fallthrough is intentional.

erwango, 2020-02-05:
@ydamigos indeed.
I've updated the record and set it as intentional/ignore

closed 2020-02-05 by erwango