zephyrproject-rtos/zephyr #22452: not driver found in can bus samples for olimexino_stm32

boonpin1975, 2020-02-04:
is the canbus supported for olimexino_stm32 ? 
reproduce :
west build -b olimexino_stm32 samples/drivers/CAN
console output : 
uart:~$ �*** Booting Zephyr OS build zephyr-v2.1.0-1628-gb45b1e393c95  ***
CAN: Device driver not found.

uart:~$ canbus
canbus - CAN commands
Subcommands:
  send    :Send a CAN frame.
           Usage: send device_name [-re] id [byte_1 byte_2 ...]
           -r Remote transmission request
            -e Extended address
  attach  :Attach a message filter and print those messages.
           Usage: attach device_name [-re] id [mask [-r]]
           -r Remote transmission request
            -e Extended address
  detach  :Detach the filter and stop receiving those messages
           Usage: detach device_name filter_id
uart:~$

erwango, 2020-02-04:
I don't have hardware to test but CAN support on olimexino_stm32 is enabled since 2302d2615b519fe2ccc1a94d33e493fc10b3953d.

Maybe @alexanderwachter you can comment on this ?

alexanderwachter, 2020-02-04:
@boonpin1975 Can you turn on the debug log-level for the CAN driver and send the logs?

alexanderwachter, 2020-02-04:
@boonpin1975 don't forget to connect the board to a CANbus that is terminated.
Otherwise, the driver does not initialize, because it needs 11 recessive bits on the bus to initialize.

boonpin1975, 2020-02-05:
hi all ,

 I am connecting the olimexino_stm32 to the komodo can solo for verification purposes , CANH -> CAN+ ( komodo ) , CANL - > CAN- (komodo) , GND -> GND ( komodo )
[image]

my prj.conf
CONFIG_CAN=y
CONFIG_CAN_INIT_PRIORITY=80
CONFIG_CAN_MAX_FILTER=5
CONFIG_CAN_STM32=y
CONFIG_CAN_1=y
CONFIG_CAN_LOG_LEVEL_DBG=y
CONFIG_SHELL=y
CONFIG_CAN_SHELL=y
CONFIG_DEVICE_SHELL=y

alexanderwachter, 2020-02-05:
Are you on the latest master?

boonpin1975, 2020-02-05:
the version is 2.1.99  . I can build the example but it can not loaded the CAN driver when connected to komodo can analyzer

alexanderwachter, 2020-02-05:
@walzert can you check if the tip of master works on your setup?

walzert, 2020-02-05:
@alexanderwachter will do that tomorrow.

alexanderwachter, 2020-02-17:
@boonpin1975 did you get your board working?

alexanderwachter, 2020-03-02:
No response from issuer since two weeks.

closed 2020-03-02 by alexanderwachter

songmaotian, 2020-03-13:
I have same problem，I open the can Log，and it's seems because it's doesn't get 11 recessive bits on the bus to leave sleep mode

erwango, 2020-03-13:
@songmaotian 
> I have same problem，I open the can Log，and it's seems because it's doesn't get 11 recessive bits on the bus to leave sleep mode

Can you open a new issue? Please provide as much as possible information on your environment and application.