zephyrproject-rtos/zephyr #22455: How to assign USB endpoint address manually in stm32f4_disco for CDC ACM class driver

darwinbeing, 2020-02-04:
I can't assign the USB endpoint address for the CDC ACM calss driver.
The below is the compile command and configuration
CLI:
west build -b stm32f4_disco zephyr/samples/subsys/usb/cdc_acm

The prj.conf file with the content:
CONFIG_STDOUT_CONSOLE=y
CONFIG_GPIO=y
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_UART_CONSOLE=y
CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_CDC_ACM=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_UART_LINE_CTRL=y
CONFIG_UART_CONSOLE_ON_DEV_NAME="CDC_ACM_0"
CONFIG_USB_DEVICE_MANUFACTURER="STMicroelectronics"
CONFIG_USB_DEVICE_PRODUCT="Zephyr/RT Virtual COM Port"
CONFIG_USB_DEVICE_VID=0x0483
CONFIG_USB_DEVICE_PID=0x5740
CONFIG_USB_DEVICE_SN="301"
# CONFIG_CDC_ACM_IAD is not set

Flashing the program and seeing the console logs on ubuntu by executing the CLI: sudo lsusb -d 0483:5740 -v.

Bus 003 Device 006: ID 0483:5740 STMicroelectronics STM32F407
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            2 Communications
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0483 STMicroelectronics
  idProduct          0x5740 STM32F407
  bcdDevice            2.01
  iManufacturer           1 STMicroelectronics
  iProduct                2 Zephyr/RT Virtual COM Port
  iSerial                 3 301
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           67
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             255
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)

The USB endpoint address obtained is as follows:
0x81 0x01 0x82

I want to get the following address by modifying the code or config.
0x82 0x01 0x81

......
   Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN      ---
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             255
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT   ---
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN   ---
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0

I tried to modifty the file.
zephyr/subsys/usb/class/cdc_acm.c

#define DEFINE_CDC_ACM_DESCR_AUTO(x, _) \
	DEFINE_CDC_ACM_DESCR(x, AUTO_EP_IN, AUTO_EP_OUT, AUTO_EP_IN);

#define DEFINE_CDC_ACM_EP_AUTO(x, _) \
	DEFINE_CDC_ACM_EP(x, AUTO_EP_IN, AUTO_EP_OUT, AUTO_EP_IN);

->
1.
#define DEFINE_CDC_ACM_DESCR_AUTO(x, _) \
	DEFINE_CDC_ACM_DESCR(x, 0x82, 0x01, 0x81);

#define DEFINE_CDC_ACM_EP_AUTO(x, _) \
	DEFINE_CDC_ACM_EP(x, 0x82, 0x01, 0x81);
2.
#define DEFINE_CDC_ACM_DESCR_AUTO(x, _) \
	DEFINE_CDC_ACM_DESCR(x, 0x81, 0x01, 0x82);

#define DEFINE_CDC_ACM_EP_AUTO(x, _) \
	DEFINE_CDC_ACM_EP(x, 0x81, 0x01, 0x82);
3.    
#define DEFINE_CDC_ACM_DESCR_AUTO(x, _) \
	DEFINE_CDC_ACM_DESCR(x, 0x01, 0x82, 0x81);

#define DEFINE_CDC_ACM_EP_AUTO(x, _) \
	DEFINE_CDC_ACM_EP(x, 0x01, 0x82, 0x81);        

I have tried lots of configs, they all fail, would you please help me out on it?

jfischer-phytec-iot, 2020-02-04:
You can not because it is not necessary. The purpose of the endpoint descriptor is to describe the configuration. The problem is your host when it needs hard-coded endpoint addresses.

darwinbeing, 2020-02-04:
I have filled hard-coded endpoint address in the the usb descriptor in the file zephyr/subsys/usb/class/cdc_acm.c, so i thought it would work，on the contrary, nothing changed, it seems that the driver re-assigned the endpoint address automatically,  where and how ?

jfischer-phytec-iot, 2020-02-04:
> it seems that the driver re-assigned the endpoint address automatically, where and how ?

The stack does it [here](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/usb_descriptor.c#L223).

darwinbeing, 2020-02-04:
Thank you for your help

 The driver re-assigns the endpoint address  in the function usb_validate_ep_cfg_data [usb_descriptor.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/usb/usb_descriptor.c#L223), after I commented the code below, I can specify the endpoint address manually.

```
		case USB_ENDPOINT_DESC:
			if (!cfg_data) {
				LOG_ERR("Uninitialized usb_cfg_data pointer, "
					"corrupted device descriptor?");
				return -1;
			}

			LOG_DBG("Endpoint descriptor %p", head);
			ep_descr = (struct usb_ep_descriptor *)head;
			if (usb_validate_ep_cfg_data(ep_descr,
						     cfg_data,
						     &requested_ep)) {
				LOG_ERR("Failed to validate endpoints");
				return -1;
			}

			break;
```

andysan, 2021-08-29:
> You can not because it is not necessary. The purpose of the endpoint descriptor is to describe the configuration. The problem is your host when it needs hard-coded endpoint addresses.

I agree that this shouldn't be an issue, good drivers should parse the interface descriptor and pick up the right endpoints. However, that's not always the case.

There are plenty of drivers in Linux and elsewhere that use hard-coded endpoint numbers. For example, I have started to implement a gs_usb-compatible USB to CAN device using Zephyr and the Linux driver simply won't work unless specific endpoint numbers are used. The driver makes no attempt parsing any interface descriptors. This is the case for the Python driver as well. Fixing the driver is of course the desirable approach, but that's not always possible.

I would argue that not being able to override automatic EP assignment is a shortcoming in Zephyr that needs to be fixed to support real-world use cases.

andysan, 2021-08-29:
I'm currently testing a patch that solves the issue for me by only enabling auto-assignment if the EP has been set to `AUTO_EP_OUT` or `AUTO_EP_IN`. Unfortunately, this means that USB descriptors need to be parsed in two passes since all of the statically assigned EP addresses need to be added to the `requested_ep` mask before automatic assignment starts. I should be able to post this patch sometime next week if this seems like the desirable behaviour (it was the behaviour I expected when implementing my device).

jfischer-no, 2021-09-20:
> There are plenty of drivers in Linux and elsewhere that use hard-coded endpoint numbers. For example, I have started to implement a gs_usb-compatible USB to CAN device using Zephyr and the Linux driver simply won't work unless specific endpoint numbers are used.

Please fix it there in Linux Kernel.

closed 2021-09-20 by jfischer-no

dawoodkhan99, 2022-04-13:
> Thank you for your help
>
> The driver re-assigns the endpoint address in the function usb_validate_ep_cfg_data [...]

@darwinbeing  I have the same issue, I want to change endpoint descriptor address but it is not working. How did you solve this by your end ?