zephyrproject-rtos/zephyr #22462: onoff: why client must be reinitialized after each transition

nordic-krch, 2020-02-04:
In notify function `client->flags` is cleared and because of that client needs to be reinitialized before each use (release, request). What is the purpose of this? I cannot find it documented and it seems redundant.

https://github.com/zephyrproject-rtos/zephyr/blob/3fba0cb5d1543e6f712d6c0414030ffd98d03b5b/lib/os/onoff.c#L106

pabigot, 2020-02-04:
That is the condition that is used by `onoff_client_fetch_result` to confirm that it's safe to read and return the stored result value.

Secondarily it allows `validate_args` to verify that the structure has been reset to a valid request, and isn't still participating in some other operation.

closed 2020-02-04 by pabigot

nordic-krch, 2020-02-05:
It is confusing to the user (like me) and I had to dig into source code to find out that flags are reset on purpose and not because something gets overwritten. It's not mentioned in the documentation as well.

I wanted to use single callback and jest pass bool variable to as user_data and control state using it and init client only once. If module has limitations like that it should be documented in the API.

pabigot, 2020-02-05:
I recall mentioning it in a comment during the discussion of how to use the API, but apparently it never was made explicit in the final documentation.  #22493.