zephyrproject-rtos/zephyr #22464: Add strdup support to core C library

SteveOss, 2020-02-04:
**Is your enhancement proposal related to a problem? Please describe.**
I am porting code to Zephyr that uses strdup, would be easier if supported in core lightweight C library

**Describe the solution you'd like**
In implementation of the POSIX strdup function (3 lines of code)

**Describe alternatives you've considered**
My own implementation of strdup

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

carlescufi, 2020-02-04:
@SteveOss would you mind sending a PR yourself?

SteveOss, 2020-02-04:
@carlescufi No problem. Will do.

SteveOss, 2020-02-05:
Fix is on PR#22491

expploitt, 2021-02-18:
Hello, this fix is where is available? It isn't still in  master

SteveOss, 2021-02-18:
@expploitt I never got sign off on the PR as the guys I was interacting with stopped interacting :( You can still find the PR by 22464, was closed as stale.

expploitt, 2021-02-18:
Well, It is interesting to have the different functions to change from string to x in zephyr. I'm trying to integrate zenoh over zephyr and there are functions like strtof or strod that are not available. :(

SteveOss, 2021-02-18:
@expploitt I was essentially doing the same sort of porting stuff. You can work round this sort of thing but you end with horrible #ifdef everywhere.

expploitt, 2021-02-18:
@SteveOss yeah, I know that when you want to integrate something in software like that is very stressful, but I think that zenoh protocol is very interesting for IoT.

SteveOss, 2021-02-18:
@expploitt Guess you know some of my ex colleagues - Angelo, Olivier and Julien :)

expploitt, 2021-02-19:
Yeah!! I spoke with them sometimes by chat for Zenoh Gitter

SteveOss, 2021-02-19:
Small world

joelguittet, 2022-12-31:
Hello,
What is the status here please?
Joel