zephyrproject-rtos/zephyr #22466: Add hx711 sensor

didi110296, 2020-02-04:
Hello, I'm new to zephyr os, I used mbed os in the past, I would like to use an hx711 module but I don't understand anything about the documentation on gpios and I also don't know if it's possible to create your own sensors.
I would like to know if it is possible to add the hx711 module directly in zephyr os, or if someone can help me create this library.

Thank you Best regards.

carlescufi, 2020-02-04:
@didi110296 it is perfectly possible to add your own sensors, but unless you have a very specific reason not to do so, I would recommend you submit it as a Pull Request and get it merged into the repository. That way any global changes will be made automatically for you and it will be checked so that it always builds even as Zephyr changes.

I recommend you find a similar sensor to the hx711 among the currently supported ones:
https://github.com/zephyrproject-rtos/zephyr/tree/master/drivers/sensor

and use that as a starting point to implement yours.

didi110296, 2020-03-14:
Hello, is it possible to use the hx711 library for linux on zephyr?
[https://github.com/torvalds/linux/blob/master/drivers/iio/adc/hx711.c](https://github.com/torvalds/linux/blob/master/drivers/iio/adc/hx711.c)

GeorgeGkinis, 2020-04-01:
I have started implementing a driver for **HX711**. 
So far it compiles as an **out-of-tree** device and also as part of **zephyr** and **stubs** are in place.

I am still debating whether to implement **out-of-tree** or do a **PR**.

No actual functionality here, just making sure I understand the **zephyr device driver model.**

There are some questions I have : 

1. Should configuration be implemented using only **devicetree** or is **Kconfig** still required?
I understand that **Kconfig** will be deprecated soon.

2. There seems to be a need to add a sensor attribute **SENSOR_ATTR_SLOPE** additionally to **SENSOR_ATTR_OFFSET** and also **SENSOR_CHAN_WEIGHT** within **zephyr/include/rivers/sensor.h** 
 The value returned from the **HX711** needs to be altered according to the following formula :  
**SENSOR_CHAN_WEIGHT**= **SENSOR_ATTR_OFFSET** + **SENSOR_ATTR_SLOPE** * **sensor_value**

3. I have seen **SENSOR_CHAN_PRIV_START** and **SENSOR_ATTR_PRIV_START** which according to documentation are sensor specific. 
How do I use them to add additional attributes and channels only for this sensor  without modifying **zephyr/include/rivers/sensor.h** ?

4. The actual attribute values of **SENSOR_ATTR_OFFSET** and **SENSOR_ATTR_SLOPE** need to be altered during run time to allow for **calibration**. How do I set initial values within **devicetree** and **Kconfig** allowing them to be altered? 
Do I just set : 

```
static struct hx711_data hx711_data = {

        .offset = CONFIG_HX711_OFFSET,
        .slope = CONFIG_HX711_SLOPE,
        .gain = CONFIG_HX711_GAIN,
        .rate = CONFIG_HX711_SAMPLING_RATE
};

struct sensor_value calibration = {
. val1 = 1,
. val2 = 5,
}
```

and modify the values through 
```
sensor_attr_set(hx711, SENSOR_CHAN_WEIGHT, SENSOR_ATTR_SLOPE, &calibration)
```
?

Can anyone from the developement team help me out with the afforementioned questions?

Thx.

MaureenHelm, 2020-04-07:
> I am still debating whether to implement **out-of-tree** or do a **PR**.
> 

A PR would be welcome and benefit other developers.
 
> 1. Should configuration be implemented using only **devicetree** or is **Kconfig** still required?
>    I understand that **Kconfig** will be deprecated soon.

No, Kconfig is not being deprecated. Please see [Devicetree versus Kconfig](https://docs.zephyrproject.org/latest/guides/dts/dt-vs-kconfig.html#devicetree-versus-kconfig) for a discussion of what belongs where.

> 2. There seems to be a need to add a sensor attribute **SENSOR_ATTR_SLOPE** additionally to **SENSOR_ATTR_OFFSET** and also **SENSOR_CHAN_WEIGHT** within **zephyr/include/rivers/sensor.h**
>    The [...]

I think you probably only need to add `SENSOR_CHAN_WEIGHT` and keep the offset/slope parameters internal to the driver.

> 3. I have seen **SENSOR_CHAN_PRIV_START** and **SENSOR_ATTR_PRIV_START** which according to documentation are sensor specific.
>    How do I use them to add additional attributes and channels only for this sensor  without modifying **zephyr/include/rivers/sensor.h** ?

@carlescufi this was added by @pdunaj in #16437. Do you have any downstream examples demonstrating its use?

> 4. The actual attribute values of **SENSOR_ATTR_OFFSET** and **SENSOR_ATTR_SLOPE** need to be altered during run time to allow for **calibration**. How do I set initial values within **devicetree** and **Kconfig** allowing them to be altered?

How are the calibrated values determined? Does the sensor have a self-calibrate routine or does the application need to provide values? I suggest using device tree properties to set their initial values.

> Hello, is it possible to use the hx711 library for linux on zephyr?
> [torvalds/linux:drivers/iio/adc/hx711.c@`master`](https://github.com/torvalds/linux/blob/master/drivers/iio/adc/hx711.c)

No, it has a different interface and a GPL license.

carlescufi, 2020-04-09:
> > 1. I have seen **SENSOR_CHAN_PRIV_START** and **SENSOR_ATTR_PRIV_START** which according to documentation are sensor specific.
> >    How do I use them to add additional attributes and channels only for this sensor  without modifying **zephyr/include/rivers/sensor.h** ?
> 
> @carlescufi this was added by @pdunaj in #16437. Do you have any downstream examples demonstrating its use?

I tried searching in our downstream but could not find any hits. @pdunaj could you comment?

pdunaj, 2020-04-09:
Hi @carlescufi , @GeorgeGkinis,

I have added this two values to open a posibility for additional attributes and channels without having to update sensor header each time. This was after I waited for channel specific to motion sensor to be added upstream. Since our channel was added we don't use private anymore but 
https://github.com/NordicPlayground/fw-nrfconnect-nrf/blob/master/include/sensor/pmw3360.h is the example of using attributes.

The idea was that anybody willing to use the driver will include the sensor header and use enums specified by it as an extension to the standard sensor api.

GeorgeGkinis, 2020-04-12:
Thanks for answering.

Unfortunately I just made a pull request and then read the above answers.

PR : [#24295](https://github.com/zephyrproject-rtos/zephyr/pull/24295)

Of course I am willing to make any changes needed.
Please review and comment.

GeorgeGkinis, 2020-04-12:
> No, Kconfig is not being deprecated. Please see Devicetree versus Kconfig for a discussion of what belongs where.

@MaureenHelm Indeed I found my answer in the Zephyr docs. 
Devicetree for HW and Kconfig for SW setup.

> I think you probably only need to add SENSOR_CHAN_WEIGHT and keep the offset/slope parameters internal to the driver.

@MaureenHelm too late :/ My PR is already on it's way.
offset/slope parameters have become SENSOR_ATTRIBUTES configured in **Kconfig** or using **sensor_set_attribute**

>How are the calibrated values determined? Does the sensor have a self-calibrate routine or does the application need to provide values? I suggest using device tree properties to set their initial values.

@MaureenHelm If the whole hardware assembly has been tested at manufacturing then **OFFSET** and **SLOPE** can be defined in **Kconfig**. In this case no run-time calibration is needed.
If not then the application (see sample app) can use the **.tare** function from the **hx711 API** to zero (calculate **OFFSET**) and then the **.calibrate** function to calculate **SLOPE** by placing a known **CONFIG_HX711_CALIBRATION_WEIGHT** on the scale. 
**OFFSET/SLOPE** have nothing to do with the hardware. They live in the **sensor_data** struct and are used by **channel_get** to transform the raw ADC measurement to actual weight.

GeorgeGkinis, 2020-04-12:
> Since our channel was added we don't use private anymore but
https://github.com/NordicPlayground/fw-nrfconnect-nrf/blob/master/include/sensor/pmw3360.h is the example of using attributes.

> The idea was that anybody willing to use the driver will include the sensor header and use enums specified by it as an extension to the standard sensor api.

@pdunaj I see that the extra attributes are defined in an **enum pmw3360_attribute** .
I already tried that way but **sensor_attr_set** requires an **enum sensor_attribute**.

I do not understand how someone could pass it as an **enum pmw3360_attribute**  to **sensor_attribute_set**.
Typecasting?

closed 2020-07-31 by github-actions[bot]