zephyrproject-rtos/zephyr #22474: boards that have Kconfig warnings on hello_world.

pabigot, 2020-02-04:
Below is a list of boards that on current master produce Kconfig warnings when asked to build hello_world.  Generated from something like:

```
for f in $(find boards -name '*_defconfig') ; do basename $f _defconfig; done \
  | sort \
  | uniq \
  | while read b ; do rm -rf build && echo $b && west build -b $b samples/hello_world 2>&1 | grep -i warning ; done \
  2>&1 | tee all-warn.log
```

```
atsamr21_xpro                 warning: IEEE802154_RF2XX (defined at drivers/ieee802154/Kconfig.rf2xx:6) was assigned the value 'y'
bbc_microbit                  warning: LOG_BUFFER_SIZE (defined at subsys/logging/Kconfig:224) was assigned the value '128' but
degu_evk                      warning: UART_SHELL_ON_DEV_NAME (defined at subsys/shell/Kconfig.backends:33) was assigned the value
em_starterkit_em7d_normal     warning: TRUSTED_EXECUTION_NONSECURE (defined at arch/Kconfig:146) was assigned the value 'y' but
esp32                         warning: ENTROPY_ESP32_RNG (defined at drivers/entropy/Kconfig.esp32:6) was assigned the value 'y'
frdm_kw41z                    warning: The choice symbol ADC_MCUX_ADC16_VREF_ALTERNATE (defined at drivers/adc/Kconfig.mcux:45)
mps2_an385                    warning: ZTEST_STACKSIZE (defined at subsys/testsuite/ztest/Kconfig:13) was assigned the value
odroid_go                     warning: ENTROPY_ESP32_RNG (defined at drivers/entropy/Kconfig.esp32:6) was assigned the value 'y'
olimex_stm32_e407             warning: TEST_RANDOM_GENERATOR (defined at subsys/random/Kconfig:8) was assigned the value 'y' but
olimex_stm32_p405             warning: TEST_RANDOM_GENERATOR (defined at subsys/random/Kconfig:8) was assigned the value 'y' but
qemu_cortex_m0                warning: LOG_BUFFER_SIZE (defined at subsys/logging/Kconfig:224) was assigned the value '128' but
qemu_xtensa                   warning: IPM_CONSOLE_STACK_SIZE (defined at drivers/console/Kconfig:169) was assigned the value
sam4s_xplained                warning: GPIO_SAM (defined at soc/arm/atmel_sam/same70/Kconfig.defconfig.series:53,
xt-sim                        warning: IPM_CONSOLE_STACK_SIZE (defined at drivers/console/Kconfig:169) was assigned the value
xt-sim_intel_s1000            warning: IPM_CONSOLE_STACK_SIZE (defined at drivers/console/Kconfig:169) was assigned the value
```

Relates to #22472 and #22417

pabigot, 2020-02-04:
@carlescufi @nashif fyi.  Not sure if all of these would trigger CI errors.  Some generate additional warnings.

nashif, 2020-02-04:
none of those would trigger errors in CI, we have lots of kconfig warnings in many of the samples, those are being fixed slowly by ulf afaik.

ulfalizer, 2020-02-06:
Those warnings are implemented in `scripts/kconfig/kconfig.py` and aren't turned into errors.

There's fewer of them than I thought at least. I'm a bit afraid of turning it into an error, because I suspect some samples might rely on turning on some option for all boards even though it only makes sense for some of them. Maybe some of those cases could be cleaned up though.

PR #22831 (2020-02-14): Fix multiple Kconfig warnings when building samples/hello_world
See commit messages.

Fixes #22474.
closed 2020-02-14 by jhedberg