zephyrproject-rtos/zephyr #22478: Bluetooth - peripheral_dis - settings_runtime_set not working

novalisek, 2020-02-04:
**Describe the bug**
Hi, I would like to change DIS model name, manufacturer etc from code, I tried to change them in peripheral_dis main.c but there is no change when scanned.

**To Reproduce**
keep 
`CONFIG_BT_GATT_DIS_MODEL="Zephyr Model"`
in prj.conf
change "Zephyr model" to "Zephyr model mod" in main.c on lines 52+53.
build and flash

**Expected behavior**
when you read DIS model characteristic, it still contains "Zephyr model", not "Zephyr model mod"

**Impact**
For me, if example is not working as expected, I don't know where to look for problem in my own code

**Environment (please complete the following information):**
 - board: nrf52840_pca10040
 - OS: W7
 - Toolchain: GNU ARM Embedded 9-2019-q4
 - Commit SHA or Version used: f21aaeffaa5ed10629915a9d198ff483865f7be9

aescolar, 2020-02-05:
CC @jhedberg

jhedberg, 2020-02-05:
Are you sure you have settings enabled? I don't see CONFIG_BT_SETTINGS=y in the peripheral_dis prj.conf.

novalisek, 2020-02-05:
Thanks, 
adding CONFIG_BT_SETTINGS=y  and using settings_load() helped

PR #22500 (2020-02-05): Bluetooth: samples: peripheral_dis: Sample requires BT_SETTINGS
The DIS service requires BT_SETTINGS otherwise it will not have a
settings handler. Instead DIS will only use Kconfig to set it's
values.

Fixes: #22478
commit 1e80efaa (2020-02-05): Bluetooth: samples: peripheral_dis: Sample requires BT_SETTINGS

The DIS service requires BT_SETTINGS otherwise it will not have a
settings handler. Instead DIS will only use Kconfig to set it's
values.

Fixes: #22478
closed 2020-02-05 by novalisek