zephyrproject-rtos/zephyr #22484: Linker error when building google_iot_mqtt sample with zephyr-sdk 0.11.1

tsvehagen, 2020-02-05:
**Describe the bug**
After upgrading to zephyr-sdk 0.11.1 I got a linker error when building `samples/net/cloud/google_iot_mqtt`. The linker complains about `_gettimeofday` and seems to be the same error as in https://github.com/zephyrproject-rtos/zephyr/issues/13384. 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf52840_pca10056 -d build/iot-sample zephyr/samples/net/cloud/google_iot_mqtt

**Screenshots or console output**
```
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: /opt/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libc_nano.a(lib_a-gettimeofdayr.o): in function `_gettimeofday_r':
gettimeofdayr.c:(.text._gettimeofday_r+0xe): undefined reference to `_gettimeofday'
collect2: error: ld returned 1 exit status
```

**Environment (please complete the following information):**
 - OS: ArchLinux
 - Toolchain: zephyr-sdk 0.11.1
 - SHA: f21aaeffaa5ed10629915a9d198ff483865f7be9

mike-scott, 2020-02-05:
To add to this, I don't see this error using the 0.10.3 SDK with basically the same setup:

- OS: ArchLinux
- Toolchain: zephyr-sdk 0.10.3
- SHA: 4fea711f (today's master)

carlescufi, 2020-02-05:
Set as high priority since this is a regression from SDK 0.10.x

SebastianBoe, 2020-02-05:
reproduced

SebastianBoe, 2020-02-05:
I'm observing that it affects 0.11.1, but not 0.11.0.

@galak : Are you aware of any changes made between 0.11.0 and 0.11.1 that could cause this?

SebastianBoe, 2020-02-05:
Strange, this is the only commit I can find in 0.11.1, and it presumably can't cause this.

https://github.com/zephyrproject-rtos/sdk-ng/commit/df84499eba3fb1c97592506a19d060113b64483d

SebastianBoe, 2020-02-05:
Ah, it could be related to the build scripts that change behaviour depending on the SDK version.

SebastianBoe, 2020-02-05:
e.g. perhaps nano is used in 0.11.1, but not 0.11.0, and perhaps nano doesn't have _gettimeofday.

SebastianBoe, 2020-02-05:
Okay, I have the answer now.

zephyr-sdk-0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libc_nano.a

has been built incorrectly.

This must be fixed by the user that built zephyr-sdk-0.11.1/arm-zephyr-eabi
@galak ?

This is a good build:

Good:

```
lib_a-gettimeofdayr.o:
         U errno
         U gettimeofday
00000000 T _gettimeofday_r

lib_a-sysgettod.o:
00000000 T gettimeofday
         U _gettimeofday_r
         U _impure_ptr

lib_a-time.o:
         U _gettimeofday_r
         U _impure_ptr
00000000 T time

```
This is a bad build:

```
lib_a-gettimeofdayr.o:
         U errno
         U _gettimeofday
00000000 T _gettimeofday_r

lib_a-sysgettod.o:
00000000 T gettimeofday
         U _gettimeofday_r
         U _impure_ptr

lib_a-time.o:
         U _gettimeofday_r
         U _impure_ptr
00000000 T time

```

Note that "lib_a-gettimeofdayr.o" should be referencing 'gettimeofday', not '_gettimeofday'.

SebastianBoe, 2020-02-05:
The good build is from 0.11.0. The bad build is from 0.11.1.

PR #22508 (2020-02-05): libc: newlibc: Implement dummy _gettimeofday to link
With the change in SDK 0.11.1 to newlib to remove
-DMISSING_SYSCALL_NAMES we now need to implement a version of
_gettimeofday.  Previously with pre SDK 0.11.1 we had a recursive mess
of _gettimeofday_r -> gettimeofday -> _gettimeofday_r.  (which are all
implemented in newlib and thus we didn't get a link error).

With SDK 0.11.1 we have: _gettimeofday_r -> _gettimeofday.  And we
should provide a version of _gettimeofday.

Fixes #22484
commit 4fb26565 (2020-02-05): libc: newlibc: Implement dummy _gettimeofday to link

With the change in SDK 0.11.1 to newlib to remove
-DMISSING_SYSCALL_NAMES we now need to implement a version of
_gettimeofday.  Previously with pre SDK 0.11.1 we had a recursive mess
of _gettimeofday_r -> gettimeofday -> _gettimeofday_r.  (which are all
implemented in newlib and thus we didn't get a link error).

With SDK 0.11.1 we have: _gettimeofday_r -> _gettimeofday.  And we
should provide a version of _gettimeofday.

Fixes #22484
closed 2020-02-05 by carlescufi