zephyrproject-rtos/zephyr #22485: Security: Provide a full-stack demo for secure boot

ceolin, 2020-02-05:
Create a full-stack demo showing capabilities of MCUboot + Zephyr. The idea is create a didactic sample to highlight security capabilities of these technologies in a realist use case. 
- [ ] Secure boot with signed image against a public key in ROM
- [ ] Upgrades
- [ ] Bootloader with revertible upgrades

carlescufi, 2020-02-05:
CC @hakonfam @SebastianBoe

SebastianBoe, 2020-02-05:
Since how to do this is documented in the MCUBoot project, perhaps the demo should also be there?

I couldn't create this sample, it would require someone with knowledge of MCUBoot,
e.g. @nvlsianpu , or a developer from MCUBoot.

nvlsianpu, 2020-02-05:
@ceolin I don't understand how should the result looks like? A sample which builds automatically mcuboot, application, does signing and what else.

SebastianBoe, 2020-02-05:
I'm not sure either, I assumed that the demo would just document how to do it, so you still need a human to follow the steps for building and signing at each step.

SebastianBoe, 2020-02-05:
A demo does not add functionality, it demonstrates existing functionality.

And automatic building of both MCUBoot and the app is not supported (upstream).

hakonfam, 2020-02-05:
One thing which is missing to have what most people associate with "secure boot" is locking the flash area which contains MCUBoot. Currently, all zephyr applications which use MCUBoot must set the `CONFIG_MPU_ALLOW_FLASH_WRITE` option, which allows writes to any location in memory. As MCUBoot does not protect itself, it is left open for modifications, and the secure boot can be avoided once an attacker is able to execute code in application space.

nvlsianpu, 2020-02-06:
^^ This is something @ioannisg declare work on.

microbuilder, 2023-07-31:
It would be good to include encrypted firmware in this workflow to make sure this actually works with the build system today, or document how to generate the encryption key and use it with the image(s).