zephyrproject-rtos/zephyr #22498: RFC: Transaction manager

nordic-krch, 2020-02-05:
## General idea
A module which executes transaction and reports back once transaction is completed. Transaction consists of fixed number of operations. Manager instance has a pointer to a function which is capable of performing base operations. Transaction contains operations and data specific to that transaction.

## Use cases
### SPI & I2C
Transaction is a set of single direction transfers like TX+RX to read the data from a sensor. Transaction contains destination details like CS pin (SPI) or device address (I2C). Using transaction can simplify Initialization of an external device since whole configuration transfers can be combined into single transaction. Fixed transactions can be stored in ROM.

Additional considerations:
- delay between operations within a transaction shall be supported since it is often required (e.g. trigger sensor measurement, wait, read result)
- pausing between operations shall be supported. Pausing is more generic than delay thus delay may be realized with pausing feature. Pausing is required if certain condition must be met before transaction can be continued (e.g. trigger operation, wait for RDY pin, read result)
- SPI: controlling multiple pins to be able to control command/data pin or control multiple devices within single transaction
- I2C: is there a case to support multiple devices in single transaction? If so then transaction data shall contain pool of address and each message should identify which address to use.
### Others
Others has been mentioned, like CAN, ADC, GPIO (e.g. bitbanging as a transaction?)
https://github.com/zephyrproject-rtos/zephyr/pull/22371#issuecomment-580765114

## Benefits
- APIs of operations used in transactions can be simplied since they need to be capable of performing only the simplest, low level operations. E.g. i2c, spi API can be limited to `async_transfer`, `configure`.
- Combination of transaction manager with queued operation manager (#22494) allows for accessing single resource to perform chain of operations on that resource.

## API draft

Transaction contains pointer to configuration, messages and optionally "paused" callback
```
struct transaction {
  void *config;
  void *msgs;
  u16_t msg_size;
  u16_t num_msgs;
  transaction_paused_fn paused_fn;
}
```

Transaction manager shall contain operation function pointer, flags, state
Transaction is performed using call like:
```
int transaction_perform(struct transaction_mngr *mngr, struct transaction *ops,
                         transaction_cb callback, void *user_date)
```

teburd, 2023-09-26:
@nordic-krch Has this been fixed with RTIO?

nordic-krch, 2023-09-29:
yes.

closed 2023-09-29 by nordic-krch