zephyrproject-rtos/zephyr #22502: after pulling latest master the usb transfer brings up warnings

StefJar, 2020-02-05:
after pulling the latest master branch. I see so much more logging on the usb and warnings. I am pretty sure that it is related to that commits:

```
commit 7a1ab248911837688ec9db75f615fe261b80e688
Author: Johann Fischer <j.fischer@phytec.de>
Date:   Thu Dec 12 17:09:44 2019 +0100    usb: rework USB transfer logging    Rework USB transfer logging.    Signed-off-by: Johann Fischer <j.fischer@phytec.de>

commit 7bd0de0c61ac279697cd9afefe7d05fa2979903c
Author: Johann Fischer <j.fischer@phytec.de>
Date:   Thu Dec 12 16:38:28 2019 +0100    usb: move USB transfer functions to appropriate file    Move USB transfer functions to appropriate file.    Signed-off-by: Johann Fischer <j.fischer@phytec.de>
```
Together with my stm32f412 using the usb as an acm I am seeing these warnings etc:

```
[00:04:37.206,000] <wrn> usb_transfer: Transfer cancelled or completed, ep 0x82
[00:04:37.215,000] <wrn> usb_transfer: Transfer cancelled or completed, ep 0x82
[00:04:37.333,000] <wrn> usb_transfer: Transfer cancelled or completed, ep 0x01
[00:04:37.341,000] <wrn> usb_transfer: Transfer cancelled or completed, ep 0x01
[00:04:37.350,000] <wrn> usb_transfer: Transfer cancelled or completed, ep 0x01
```
the usb acm seems to work like before.

@jfischer-phytec-iot can you pls have a check on that?

jfischer-phytec-iot, 2020-02-05:
@StefJar I will check it asap. We did not log it earlier and I did not change any logic in the transfers, so it probably just improper log level.

sslupsky, 2020-04-01:
Has this issue been investigated?

I updated my zephyr from 2.1.0 to 2.2.0 and USB is broken.  I am seeing the same thing reported here on sam0:
```
[00:00:19.518,615] <wrn> usb_transfer: Transfer cancelled or completed, ep 0x01
[00:00:19.518,981] <wrn> usb_transfer: Transfer cancelled or completed, ep 0x82
[00:00:19.519,165] <wrn> usb_transfer: Transfer cancelled or completed, ep 0x82
[00:00:19.519,348] <wrn> usb_transfer: Transfer cancelled or completed, ep 0x82
[00:00:19.519,714] <wrn> usb_transfer: Transfer cancelled or completed, ep 0x82
[00:00:19.520,080] <wrn> usb_transfer: Transfer cancelled or completed, ep 0x82
[00:00:19.520,446] <wrn> usb_transfer: Transfer cancelled or completed, ep 0x82
```

usb_transfer is spewing out hundreds of log messages per second.  This issue was not present in 2.1.0.

sslupsky, 2020-04-01:
Ok, I see what you mean about the log level.  There is "new" log warning added that is causing this issue.  So, what is causing this?  Is this some kind of error?

jfischer-phytec-iot, 2020-04-16:
> So, what is causing this? Is this some kind of error?

Just a wrong log level, probably accidentally changed for debugging.

PR #24415 (2020-04-17): usb: transfer: fix too high log level
Lower log level in a debug message.

Fixes: #22502
commit 2c9ae5f8 (2020-04-17): usb: transfer: fix too high log level

Lower log level in a debug message.

Fixes: #22502
closed 2020-04-17 by MaureenHelm