zephyrproject-rtos/zephyr #22522: GPIO test code tests/drivers/gpio/gpio_basic_api does not compile for microchip board mec15xxevb_assy6853

ashaannajoy, 2020-02-06:
**Describe the bug**
Tried to compile the standalone gpio test code for microchip and it gives compilation error.
What have you tried to diagnose or workaround this issue?
Adding the line #define DT_ALIAS_GPIO_0_LABEL DT_MICROCHIP_XEC_GPIO_40081000_LABEL
in the file zephyr/tests/drivers/gpio/gpio_basic_api/src/test_gpio.h solves the issue. But this has to be added for all ports and in the proper way in the zephyr stack.

**To Reproduce**
Steps to reproduce the behavior:
1.	Cd zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api/
2.	West build -p -b mec15xxevb_assy6853 .
3.     See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Build should succeed. It affect gpio testing for microchip.

**Screenshots or console output**
In file included from ../src/test_callback_manage.c:15:
../src/test_gpio.h:22:2: error: #error Unsupported board
 #error Unsupported board
  ^~~~~
../src/test_callback_manage.c: In function 'test_callback_add_remove':
../src/test_callback_manage.c:84:42: error: 'DEV_NAME' undeclared (first use in this function)
  struct device *dev = device_get_binding(DEV_NAME);
                                          ^~~~~~~~
../src/test_callback_manage.c:84:42: note: each undeclared identifier is reported only once for each function it appears in
../src/test_callback_manage.c: In function 'test_callback_self_remove':
../src/test_callback_manage.c:126:42: error: 'DEV_NAME' undeclared (first use in this function)
  struct device *dev = device_get_binding(DEV_NAME);
                                          ^~~~~~~~
../src/test_callback_manage.c: In function 'test_callback_enable_disable':
../src/test_callback_manage.c:171:42: error: 'DEV_NAME' undeclared (first use in this function)
  struct device *dev = device_get_binding(DEV_NAME);
                                          ^~~~~~~~
ninja: build stopped: subcommand failed.

**Environment (please complete the following information):**
1.	OS : Ubuntu LTS 18.04
2.	Toolchain: Zephyr SDK : 0.10.3
3.	Zephyr version: v2.1.0-1048-gdbdfd2995b2b 

**Additional context**

jenmwms, 2020-02-07:
- There is an updated Zephyr SDK (0.11.1). 
- I'm not sure if this helps/relates to this, but there was recent activity on the GPIO API that may have impacted some users despite best efforts to ease the transition. Please see more information in the TOPIC-GPIO project page on that effort to convert to the new GPIO API: https://github.com/zephyrproject-rtos/zephyr/issues/20017. More specifically, https://github.com/zephyrproject-rtos/zephyr/pull/19555/commits/b3f835377d37d5909b22e7bc2fc8a30f6f1881d4

pabigot, 2020-02-11:
See #22639 which documents what needs to be done to build this test.

dcpleung, 2020-02-12:
I have no problem building it at current master.

The git commit the bug pointed to was before the topic-gpio branch was merged into master. In the topic-gpio branch, an overlay was added for the `mec15xxevb_assy6853` board (which is now in master).

Please try building again and see if it is still a problem on your side.

ashaannajoy, 2020-02-12:
Yes the build issue for microchip is fixed with the latest master.
Closing the issue

PR #22639 (2020-02-12): tests: drivers: gpio_basic_api: add README
Successful execution of this test requires a board-specific overlay to
identify the test pins, and that the test pins be shorted together.
Document this requirement.

Fixes #22622
Fixes #22522
closed 2020-02-12 by ashaannajoy