zephyrproject-rtos/zephyr #22525: stm32f7xx.h: No such file or directory

adrianalin, 2020-02-06:
Fail to build `samples/hello_world` for `stm32f769i_disco` target.

**To Reproduce**
Steps to reproduce the behavior:
1. `cd samples/hellow_world`
2. `cmake -B build -DBOARD=stm32f769i_disco .`
3. `cd build`
4. `make`
5. see error

**Expected behavior**
Build successfully.

**Impact**
Cannot run examples.

**Screenshots or console output**
```
Scanning dependencies of target offsets
[  7%] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
In file included from /home/adrian/stm32f7/zephyr/include/arch/arm/aarch32/cortex_m/cmsis.h:17,
                 from /home/adrian/stm32f7/zephyr/arch/arm/include/aarch32/cortex_m/stack.h:23,
                 from /home/adrian/stm32f7/zephyr/arch/arm/include/kernel_arch_data.h:33,
                 from /home/adrian/stm32f7/zephyr/arch/arm/core/offsets/offsets_aarch32.c:26,
                 from /home/adrian/stm32f7/zephyr/arch/arm/core/offsets/offsets.c:12:
/home/adrian/stm32f7/zephyr/soc/arm/st_stm32/stm32f7/soc.h:24:10: fatal error: stm32f7xx.h: No such file or directory
   24 | #include <stm32f7xx.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [zephyr/CMakeFiles/offsets.dir/build.make:63: zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:760: zephyr/CMakeFiles/offsets.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

**Environment (please complete the following information):**
 - Lubuntu `5.0.0-38-generic #41-Ubuntu`
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: `9d1614477efb45455021bf1eda4f74598579927b`

jfischer-phytec-iot, 2020-02-06:
I bet `west update` will resolve the issue.

adrianalin, 2020-02-06:
This also helped: https://docs.zephyrproject.org/latest/guides/modules.html and https://github.com/zephyrproject-rtos/hal_stm32

closed 2020-02-06 by adrianalin