zephyrproject-rtos/zephyr #22534: sanitycheck qemu_x86_coveragecoverage problem with SDK 0.11

pabigot, 2020-02-06:
sanitycheck --platform qemu_x86_coverage -C -T tests/lib/onoff/

locally is producing:

```
INFO    - JOBS: 32
INFO    - Building initial testcase list...
INFO    - 1 test configurations selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0
INFO    - 1 of 1 tests passed (100.00%), 0 failed, 0 skipped with 0 warnings in 36.72 seconds
INFO    - In total 1 test cases were executed on 1 out of total 228 platforms (0.44%)
INFO    - Generating coverage files...
geninfo: ERROR: need tool /usr/local/zephyr-sdk/i586-zephyr-elf/bin/i586-zephyr-elf-gcov!
lcov: ERROR: cannot read file /mnt/nordic/zp/zephyr/sanity-out/coverage.info!
lcov: ERROR: cannot read file /mnt/nordic/zp/zephyr/sanity-out/coverage.info!
lcov: ERROR: cannot read file /mnt/nordic/zp/zephyr/sanity-out/coverage.info!
lcov: ERROR: cannot read file /mnt/nordic/zp/zephyr/sanity-out/coverage.info!
genhtml: ERROR: cannot read file /mnt/nordic/zp/zephyr/sanity-out/coverage.info!
```

`i586-zephyr-elf-gcov` was last seen in SDK 0.10.x, though `x86_64-zephyr-elf-gcov` appears to be in SDK 0.11.x.

andrewboie, 2020-02-06:
I should be able to fix this, it doesn't matter much which GCOV binary you use as long as it's from the same build of GCC.

Why only this test is failing baffles me...

andrewboie, 2020-02-06:
Huh, if I use x86_64 gcov, it spews many errors. These looks vaguely familiar and prior fixes required an sdk update, maybe the gcov x64 isn't built correctly?

```
$ sanitycheck --outdir /dev/shm/sanity-out -p qemu_x86_coverage -C -T tests/lib/onoff/
Renaming output directory to /dev/shm/sanity-out.2
INFO    - JOBS: 32
INFO    - Building initial testcase list...
INFO    - 1 test configurations selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    0
INFO    - 1 of 1 tests passed (100.00%), 0 failed, 0 skipped with 0 warnings in 33.61 seconds
INFO    - In total 1 test cases were executed on 1 out of total 229 platforms (0.44%)
INFO    - Generating coverage files...
geninfo: WARNING: /dev/shm/sanity-out/qemu_x86_coverage/tests/lib/onoff/libraries.onoff/zephyr/kernel/CMakeFiles/kernel.dir/device.c.gcno: Overlong record at end of file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#include#kernel.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#kernel#device.c.gcov in .gcno file, skipping file!
geninfo: WARNING: /dev/shm/sanity-out/qemu_x86_coverage/tests/lib/onoff/libraries.onoff/zephyr/kernel/CMakeFiles/kernel.dir/errno.c.gcno: Overlong record at end of file!
geninfo: WARNING: cannot find an entry for #dev#shm#sanity-out#qemu_x86_coverage#tests#lib#onoff#libraries.onoff#zephyr#include#generated#syscalls#errno_private.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#kernel#errno.c.gcov in .gcno file, skipping file!
geninfo: WARNING: /dev/shm/sanity-out/qemu_x86_coverage/tests/lib/onoff/libraries.onoff/zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.gcno: Overlong record at end of file!
geninfo: WARNING: cannot find an entry for #dev#shm#sanity-out#qemu_x86_coverage#tests#lib#onoff#libraries.onoff#zephyr#include#generated#syscalls#kernel.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#include#arch#x86#ia32#arch.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#include#logging#log.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#include#logging#log_core.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#include#syscall.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#kernel#fatal.c.gcov in .gcno file, skipping file!
geninfo: WARNING: /dev/shm/sanity-out/qemu_x86_coverage/tests/lib/onoff/libraries.onoff/zephyr/kernel/CMakeFiles/kernel.dir/idle.c.gcno: Overlong record at end of file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#include#arch#x86#ia32#arch.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#include#kernel.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#kernel#idle.c.gcov in .gcno file, skipping file!
geninfo: WARNING: /dev/shm/sanity-out/qemu_x86_coverage/tests/lib/onoff/libraries.onoff/zephyr/kernel/CMakeFiles/kernel.dir/init.c.gcno: Overlong record at end of file!
geninfo: WARNING: cannot find an entry for #dev#shm#sanity-out#qemu_x86_coverage#tests#lib#onoff#libraries.onoff#zephyr#include#generated#syscalls#kernel.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#arch#x86#include#ia32#kernel_arch_func.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#include#arch#x86#arch.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#include#arch#x86#ia32#arch.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#include#sys#dlist.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#kernel#include#ksched.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#kernel#include#kswap.h.gcov in .gcno file, skipping file!
geninfo: WARNING: cannot find an entry for #home#apboie#projects#zephyr#zephyr#kernel#init.c.gcov
```

andrewboie, 2020-02-06:
@nashif patching sanitycheck is trivial but the x64 gcov binary doesn't seem to work.

```
diff --git a/scripts/sanitycheck b/scripts/sanitycheck
index 13f9eecd63..8599e9bc3c 100755
--- a/scripts/sanitycheck
+++ b/scripts/sanitycheck
@@ -4361,7 +4361,7 @@ def main():
                 options.gcov_tool = "gcov"
             else:
                 options.gcov_tool = os.path.join(os.environ["ZEPHYR_SDK_INSTALL_DIR"],
-                                                 "i586-zephyr-elf/bin/i586-zephyr-elf-gcov")
+                                                 "x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcov")
 
         logger.info("Generating coverage files...")
         coverage_tool = CoverageTool.factory(options.coverage_tool)
```

nashif, 2020-02-07:
what lcov version do you guys have?

andrewboie, 2020-02-07:
@nashif I'm on Ubuntu 18.04 which has LCOV 1.13. That's probably the issue, I'll upgrade and re-test.

PR #22609 (2020-02-07): sanitycheck: point to correct gcov path
The SDK has a new path for x86 gcov, point to the new binary. Also, do
not clean artifacts when running coverage.

Fixes #22534
closed 2020-02-07 by nashif