zephyrproject-rtos/zephyr #22535: drivers: lora: Make the SX1276 driver independent of loramac module

KubaFYI, 2020-02-06:
### Problem description

The Semtech sx1276 driver currently selects the incomplete loramac-node module to be built with it. This is unnecessary as users might wish to use just the driver without the entire LoRaMAC stack. Besides, currently the loramac-node contains it's own duplicate of the sx1276 driver code which is confusing.

### Proposed change

Make it so that loramac-node (through `HAS_SEMTECH_LORAMAC`) requires the sx1276 driver and not the other way around.

KubaFYI, 2020-02-06:
@Mani-Sadhasivam could you please let me know if that makes sense?

Mani-Sadhasivam, 2020-02-06:
@KubaFYI The SX1276 driver in Zephyr is a shim driver for the one available in loramac-node repo. During my first attempt to add LoRa support, I made the SX1276 independent of the LoRAMAC stack but that was NACKed and folks asked me to use the existing work available in loramac-node. Hence, I came up with this shim driver for Zephyr.

So the Zephyr SX1276 driver should depend on the loramac-node repo. Your proposed change won't work as loramac-node repo doesn't have any dependency with Zephyr.

KubaFYI, 2020-02-06:
Thanks, understood. 

May I ask what's the rationale behind taking the entirety of loramac-node code and including it as a module rather than porting it as a driver + library? I'm in a process of building a Lorawan node on zephyr and I'm trying to understand what's the best way for me to go forward.

Mani-Sadhasivam, 2020-02-06:
> May I ask what's the rationale behind taking the entirety of loramac-node code and including it as a module rather than porting it as a driver + library?

I'm not sure if I understand your question. Currently loramac-node is built as a library. Not all the configurations in the loramac-node is enabled atm. You can look [here](https://github.com/zephyrproject-rtos/loramac-node/blob/master/zephyr/CMakeLists.txt).

> I'm in a process of building a Lorawan node on zephyr and I'm trying to understand what's the best way for me to go forward.

Cool. I'm also working on LoRAWAN support in parallel. Let me know if you want to join the effort.

KubaFYI, 2020-02-06:
> I'm not sure if I understand your question.

I suppose I meant why is it a module instead of just cherry-picked files put in `zephyr/lib` and `zephyr/drivers/lora`, but that just comes from me not being that familiar with zephyr's build process and project structure.

> Cool. I'm also working on LoRAWAN support in parallel. Let me know if you want to join the effort.

Grand, I'll ping you an email.

closed 2020-02-06 by KubaFYI