zephyrproject-rtos/zephyr #22539: bt_gatt: unable to save SC: no cfg left

chris-makaio, 2020-02-06:
**Describe the bug**
I'm regularly seeing the log err
> bt_gatt: unable to save SC: no cfg left

on a custom nRF52840 board with current master.
Sometimes there's also an assert:

> ASSERTION FAIL [status == 0 || param == ull_update_mark_get() || param == ull_disable_mark_get()] @ ZEPHYR_BASE/subsys/bluetooth[00:06:43.624,938] <err> os: r0/a1:  0x00000003  r1/a2:  0x20001494  r2/a3:  0x00000000
> [00:06:43.624,938] <err> os: r3/a4:  0x00000004 r12/ip:  0x00000000 r14/lr:  0x000232a1
> [00:06:43.624,938] <err> os:  xpsr:  0x6100001b
> [00:06:43.624,938] <err> os: Faulting instruction address (r15/pc): 0x000232ac
> [00:06:43.624,938] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
> [00:06:43.624,938] <err> os: Fault during interrupt handling
> 
> [00:06:43.624,969] <err> os: Current thread: 0x20002af4 (BT RX)
> [00:06:53.336,456] <err> errhandler: Halting system

and I have no idea where I should look at.

I tried to disable GATT Caching and I'm already using a single connection concept (BT_MAX_PAIRED = 1)

**To Reproduce**
It only happens when a "new" device connects. "New" in terms of a new MAC. The reason we stumbled upon this is the fact that iOS randomly changes the MAC address for the iDevice which led to errors. 
So, this works:
- boot Zephyr
- connect via Lightblue or any other BLE app
- disconnect
- reconnect
- repeat as many times as you want 

This doesn't work:
- boot Zephyr
- connect via iPhone (again, Lightblue etc)
- disconnect
- enable and disable flight mode to get a new MAC for your iPhone
- connect
- discussed issues come up

chris-makaio, 2020-02-06:
Maybe related to #22514

aescolar, 2020-02-06:
CC @jhedberg @joerchan

jhedberg, 2020-02-06:
Added both @cvinayak and @Vudentz as assignees, since while the Service Changed warning is host-related, the assert looks like it's controller-related.

chris-makaio, 2020-02-06:
here's some more log output:

> [00:24:58.253,234] <dbg> bt_gatt.sc_save: peer 58:dd:00:e0:85:9c (random) start 0x0000 end 0x0000
> [00:24:58.253,326] <dbg> bt_gatt.find_sc_cfg: id: 0, addr: 58:dd:00:e0:85:9c (random)
> [00:24:58.253,417] <dbg> bt_gatt.find_sc_cfg: id: 0, addr: 00:00:00:00:00:00 (public)
> [00:24:58.253,448] <err> bt_gatt: unable to save SC: no cfg left

Obviously as in #22514 the sc_cfg is never flushed, it seems

commit 4edfd45c (2020-02-14): Bluetooth: GATT: Fix not clearing SC when disconnecting

SC config data is no longer stored within the CCC config itself
therefore it must be cleared separately.

Fixes #22539
closed 2020-02-14 by jhedberg