zephyrproject-rtos/zephyr #22542: GEN_ABSOLUTE_SYM cannot handle value larger than INT_MAX on qemu_x86_64

dcpleung, 2020-02-06:
If a kconfig (e.g. `CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC` is defined with a value > +2147483647, it fails to compile with following error:

```
In file included from ../include/toolchain.h:39,
                 from zephyr/misc/generated/configs.c:7:
zephyr/misc/generated/configs.c: In function '_ConfigAbsSyms':
../include/toolchain/gcc.h:404:2: error: invalid 'asm': operand is not a condition code, invalid operand code 'c'
  404 |  __asm__(".globl\t" #name "\n\t.equ\t" #name \
      |  ^~~~~~~
zephyr/misc/generated/configs.c:14:1: note: in expansion of macro 'GEN_ABSOLUTE_SYM'
   14 | GEN_ABSOLUTE_SYM(CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC, 2147483648);
      | ^~~~~~~~~~~~~~~~
```

I haven't tried negative maximum but it is reasonable to believe that it would also fail to compile.

andrewboie, 2020-02-11:
I'll take a look.

andrewboie, 2020-02-11:
The "c" modifier:

```
c | Require a constant operand and print the constant expression with no punctuation.
-- | --
```

From https://gcc.gnu.org/onlinedocs/gcc/Extended-Asm.html in 6.47.2.8 x86 Operand Modifiers

Seems like it is not a fan of constants that don't fit in a 32-bit integer? hmmm

dcpleung, 2020-02-11:
Note that this fails only on `qemu_x86_64`, but works on `qemu_x86`.

andrewboie, 2020-02-11:
This boils down to the following GCC inline asm code:
```
__asm__(".globl CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC\n\t"
        ".equ CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC,%c0\n\t"
        ".type CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC,@object"
		: : "n"(3000000000));
```

We are trying to set the value with `.equ` but it is not happy with the combination of `%c0` for the first operand, modified with `c`, and the operand constraint is itself `n`.

The fact that this works fine with x86 32-bit is truly baffling.

andrewboie, 2020-02-11:
I might have to ask in the GCC mailing list about this one. The documentation for all of this is so far not helping.

andrewboie, 2020-02-11:
> Note that this fails only on `qemu_x86_64`, but works on `qemu_x86`.

It passes if the value is less than UINT_MAX, if it's bigger than that we get an overflow warning instead, it still builds but the value is truncated:

```
/home/apboie/projects/zephyr/zephyr/tests/kernel/mem_protect/userspace/out/zephyr/include/generated/autoconf.h:3:44: warning: overflow in conversion from 'long long int' to 'int' changes value from '9000000000' to '410065408' [-Woverflow]
    3 | #define CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC 9000000000
      |                                            ^~~~~~~~~~
/home/apboie/projects/zephyr/zephyr/kernel/timeout.c:33:33: note: in expansion of macro 'CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC'
   33 | int z_clock_hw_cycles_per_sec = CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC;
      |                      
```

andrewboie, 2020-02-11:
So what I'm seeing here is as follows:

* This mechanism likely does not work for any arch if the value exceeds UINT_MAX. In addition, on i386 it seems to fail silently. We might need some other macro type for values that can get this large.
* This mechanism causes an error on x86-64 if the value exceeds INT_MAX, unlike other arches.

For the former, we probably should open another bug if we're expecting to start dealing with Kconfigs that have values larger than UINT_MAX. We might consider other alternatives such as having CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC be expressed in terms of KHz instead of Hz, but that may or may not be feasible.

For the latter, I've sent a message to gcc-help, maybe someone knows a workaround.

PR #22764 (2020-02-13): x86: fix large Kconfig values on 64-bit
Use the special x86 operand modifier 'p' to print the raw value.
This fixes an issue on x86-64 where errors were generated
if a constant larger than INT_MAX was used.

Values larger than UINT_MAX are still unsupported (on any arch).

Fixes: #22542
commit f1b1332c (2020-02-13): x86: fix large Kconfig values on 64-bit

Use the special x86 operand modifier 'p' to print the raw value.
This fixes an issue on x86-64 where errors were generated
if a constant larger than INT_MAX was used.

Values larger than UINT_MAX are still unsupported (on any arch).

Fixes: #22542
closed 2020-02-13 by jhedberg