zephyrproject-rtos/zephyr #22554: Add support to device tree generation support for DT_PATH_<path>_<FOO> generation

galak, 2020-02-06:
Add support to scripts/dts/gen_defines.py to generate device tree path based defines.  We should utilize these defines as the canonical defines and phase out the current 'ident' defines which are roughly COMPAT+UNIT_ADDR plus some logic if we are on a 'matching' bus.

So for something like:

```
        soc {
                flash-controller@40020000 {
                        flash0: flash@0 {
                                reg = < 0x0 0x100000 >;
                                };
                        };
                };
        };
```

The path is:
`/soc/flash-controller@40020000/flash@0`

and we should generate:

`#define DT_PATH_SOC_FLASH_CONTROLLER_40020000_BASE_ADDRESS 0x40020000`

galak, 2020-02-06:
POC of the idea is here: #22553

pabigot, 2020-02-07:
We should make clear that from a node:
```
[label:] node-name[@unit-address] {
  [properties definitions]
  [child nodes]
};
```
the path element is constructed first from a normalized (for capitalization and symbol-to-underscore) representation of the node name.  If the unit address is present (and matches the first address in the `reg` property, as required) this is immediately followed by an underscore and the lower-case hex representation of the unit address with leading zeros removed.  The path elements are then joined with underscores.

Why: According to [the specification](https://github.com/devicetree-org/devicetree-specification/blob/master/source/devicetree-basics.rst#path-names) the unit address can be omitted if the resulting path is unique, but we can't assume that will be true if the devicetree source is changed, so we should disallow that omission.

As a separate change it would be nice if we transitioned to `UNIT_ADDRESS` rather than `BASE_ADDRESS` as the pseudo-property identifier corresponding to the unit address.

mbolivar, 2020-02-07:
> As a separate change it would be nice if we transitioned to `UNIT_ADDRESS` rather than `BASE_ADDRESS` as the pseudo-property identifier corresponding to the unit address.

This is actually from the reg property, not unit address.

pabigot, 2020-02-10:
OK, after some investigation I've updated [my proposal](https://github.com/zephyrproject-rtos/zephyr/issues/22554#issuecomment-583446523).

The point of that proposal is to make clear that eliding the unit address from the canonical representation of the path is not permitted even if a path that used only node-name components is unique within a given devicetree.

Which means for this:
```
/dts-v1/;

/ {
  #address-cells = <2>;
  #size-cells = <1>;
  a@0,1 {
    reg = <0 1 2>;
    tag = "first";
  };
  b@0,2 {
    reg = <0 2 3>;
    tag = "second";
  };
};
```

the generated paths must be `DT_PATH_A_0_1` and `DT_PATH_B_0_2` rather than `DT_PATH_A` and `DT_PATH_B`.

To the point of generating  `_BASE_ADDRESS`:  "base address" is not a technical term in devicetree, but "unit address" is.  We should move towards terminology that is consistent with the specification.

mbolivar, 2020-02-10:
> To the point of generating `_BASE_ADDRESS`: "base address" is not a technical term in devicetree, but "unit address" is. We should move towards terminology that is consistent with the specification.

I agree with the general principle; I disagree where it pertains to this example.

Since the BASE_ADDRESS macros are not generated from node unit addresses, a name with UNIT_ADDRESS in it would be at best misleading.

galak, 2020-02-10:
> Since the BASE_ADDRESS macros are not generated from node unit addresses, a name with UNIT_ADDRESS in it would be at best misleading.

I agree with @mbolivar on this.

I'm not sure why we didn't just call it `REG` to match the property name.

pabigot, 2020-02-10:
I think we're not sharing an understanding.  From what I can tell the unit address of a node **is** the first address in the `reg` value.

You can see references to "unit address" in, for example, [interrupt-map](https://github.com/devicetree-org/devicetree-specification/blob/master/source/devicetree-basics.rst#interrupt-map).

That the devicetree source requires presence of the unit address as a `unit-address` token in the node identifier is probably just to guarantee disambiguation.  I don't see it as being anything other than a value derived from the `reg` property address cells.

From https://github.com/zephyrproject-rtos/zephyr/blob/master/scripts/dts/gen_defines.py#L211 it seems the tooling generates `_BASE_ADDRESS` from the reg address, and `_SIZE` from the size.  You can't call it `_REG`; you could call it `_REG_ADDRESS` and `_REG_SIZE`.  But if you're just going to provide an address, `_UNIT_ADDRESS` at least seems to match the specification.  `_BASE_ADDRESS` appears to be a Zephyr construction.

'Nuf said; probably too late to change anyway.

mbolivar, 2020-02-11:
> I'm not sure why we didn't just call it `REG` to match the property name.

I have some general ideas about naming that are along these lines.

I think `_PROP_<PROPNAME>_` is a reasonable "infix" requirement for macros derived from properties, where `PROPNAME` is the property name converted to a C identifier with the existing rules.

So that would mean `DT_<NAMESPACE>_PROP_REG_BASE_ADDRESS` is the final macro, where `<NAMESPACE>` is something like `INST_<N>_<COMPAT>`, `<ALIAS>_<ALIASNAME>`, etc.

> From what I can tell the unit address of a node **is** the first address in the `reg` value.

It's possible not to obey this convention, though dtc warns about it.

closed 2020-03-24 by galak