zephyrproject-rtos/zephyr #22555: Add support to device tree generation support for DT_NODE_LABEL_<node-label>_<FOO> generation

galak, 2020-02-06:
Add support to scripts/dts/gen_defines.py to generate device tree node label based defines.  These should be of the form `DT_NODEL_LABEL_<node-label>_<FOO>`

So for something like:

        soc {
                flash-controller@40020000 {
                        cool_label_name: flash0: flash@0 {
                                flash0_reg: reg = < 0x0 0x100000 >;
                                };
                        };
                };
        };

```
/* assuming Issue #22554 - DT_PATH_ */
#define DT_NODE_LABEL_FLASH0_BASE_ADDRESS DT_PATH_SOC_FLASH_CONTROLLER_40020000_BASE_ADDRESS
#define DT_NODE_LABEL_COOL_LABEL_NAME_BASE_ADDRESS DT_PATH_SOC_FLASH_CONTROLLER_40020000_BASE_ADDRESS
```
NOTE: We should only generate labels for node's at this time.  So the label `flash0_reg` on a property should NOT generate a define.  In the future if we need this we should call it `DT_PROP_LABEL_...`

galak, 2020-02-06:
Need to document as part of this work:
* node labels should ideally get used rarely in the code, should be for true cases that `DT_INST` defines can't be used because you need to know the actual SoC/HW instance
* If a label is used for SoC identification it should match the SoC Vendor documentation for that device.

erwango, 2020-02-07:
Having a `DT_NODEL_LABEL_<node-label>` will be required to enable used as;
```
#ifdef DT_NODEL_LABEL_<node-label>
#endif
```

pabigot, 2020-02-07:
I would strongly prefer `DT_NODELABEL_` as the prefix rather than introduce what appear to be syntactic subelements within the namespace identifier.

closed 2020-03-24 by galak