zephyrproject-rtos/zephyr #22559: crash in semaphore tests on ARC nsim_em and nsim_sem

andrewboie, 2020-02-06:
**Describe the bug**
Seeing some crashes or hangs in tests on ARC. This seems to be a new regression.

nsim_em/
tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem
```
*** Booting Zephyr OS build zephyr-v2.1.0-2101-g3a2f34b2f594  ***
Running test suite test_sys_sem
===================================================================
starting test - test_basic_sem_test
PASS - test_basic_sem_test
===================================================================
starting test - test_simple_sem_from_isr
PASS - test_simple_sem_from_isr
===================================================================
starting test - test_sem_take_timeout_isr
zephyr/CMakeFiles/run.dir/build.make:57: recipe for target 'zephyr/CMakeFiles/run' failed
CMakeFiles/Makefile2:490: recipe for target 'zephyr/CMakeFiles/run.dir/all' failed
CMakeFiles/Makefile2:497: recipe for target 'zephyr/CMakeFiles/run.dir/rule' failed
Makefile:222: recipe for target 'run' failed
```

nsim_sem/tests/kernel/mem_protect/sys_sem/kernel.memory_protection.sys_sem/handler.log
```
Console now belongs to UART, hit CRTL-] to return to simulator.
*** Booting Zephyr OS build zephyr-v2.1.0-2101-g3a2f34b2f594  ***
Running test suite test_sys_sem
===================================================================
starting test - test_basic_sem_test
PASS - test_basic_sem_test
===================================================================
starting test - test_simple_sem_from_isr
PASS - test_simple_sem_from_isr
===================================================================
starting test - test_sem_take_timeout_isr
E: ***** Exception vector: 0x6, cause code: 0x1, parameter 0x24
E: Address 0x0
E: EV_ProtV
E: Memory read protection violation (Secure MPU)
E: Faulting instruction address = 0x3290
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x8000031c (unknown)
E: Halting system
Console now belongs to simulator.
[100%] Built target run
```

nsim_sem/tests/kernel/semaphore/semaphore/kernel.semaphore
```
Scanning dependencies of target run
Console now belongs to UART, hit CRTL-] to return to simulator.
*** Booting Zephyr OS build zephyr-v2.1.0-2101-g3a2f34b2f594  ***
Running test suite test_semaphore
===================================================================
starting test - test_k_sema_init
PASS - test_k_sema_init
===================================================================
starting test - test_sema_thread2thread
PASS - test_sema_thread2thread
===================================================================
starting test - test_sema_thread2isr
PASS - test_sema_thread2isr
===================================================================
starting test - test_sema_reset
PASS - test_sema_reset
===================================================================
starting test - test_sema_count_get
PASS - test_sema_count_get
===================================================================
starting test - test_simple_sem_from_isr
PASS - test_simple_sem_from_isr
===================================================================
starting test - test_simple_sem_from_task
PASS - test_simple_sem_from_task
===================================================================
starting test - test_sem_take_no_wait
PASS - test_sem_take_no_wait
===================================================================
starting test - test_sem_take_no_wait_fails
PASS - test_sem_take_no_wait_fails
===================================================================
starting test - test_sem_take_timeout_fails
PASS - test_sem_take_timeout_fails
===================================================================
starting test - test_sem_take_timeout
PASS - test_sem_take_timeout
===================================================================
starting test - test_sem_take_timeout_forever
PASS - test_sem_take_timeout_forever
===================================================================
starting test - test_sem_take_timeout_isr
E: ***** Exception vector: 0x6, cause code: 0x1, parameter 0x24
E: Address 0x0
E: EV_ProtV
E: Memory read protection violation (Secure MPU)
E: Faulting instruction address = 0x37ec
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x800004a4 (unknown)
E: Halting system
Console now belongs to simulator.
[100%] Built target run
```

jhedberg, 2020-02-12:
@andrewboie could you retest with latest master now that #22608 has been merged?

andrewboie, 2020-02-12:
This is now fixed.

closed 2020-02-12 by andrewboie