zephyrproject-rtos/zephyr #22561: tests/kernel/mem_protect/syscalls fails test_string_nlen on nsim_sem

andrewboie, 2020-02-06:
**Describe the bug**
```
*** Booting Zephyr OS build zephyr-v2.1.0-2104-g46b3af124a93  ***
Running test suite syscalls
===================================================================
starting test - test_string_nlen

    Assertion failed at ZEPHYR_BASE/tests/kernel/mem_protect/syscalls/src/main.c:174: test_string_nlen: (err not equal to -1)
nonsense string address did not fault
FAIL - test_string_nlen
```

On further inspection, it appears that on ARC, if you are in supervisor mode, it seems you can read the memory address 0xFFFFFFF0 without generating a fault.

This only affects nsim_sem and not nsim_em.

@vonhust @abrodkin is this expected? Is this a valid address in the memory map?

I could change the test case to only run the failing scenario from user mode, but I was hoping to exercise the exception recovery code directly.

ruuddw, 2020-02-08:
This could be expected behaviour, the nsim simulator default behavior is to simulate memory on the full address space. In kernel mode, there are no MPU restrictions either, so I expect 0xFFFFFFF0 is simply a valid memory address. I believe this nsim behavior can be changed, probably cleaner to only model the memory defined in the dts file.

andrewboie, 2020-02-11:
> This could be expected behaviour, the nsim simulator default behavior is to simulate memory on the full address space. In kernel mode, there are no MPU restrictions either, so I expect 0xFFFFFFF0 is [...]

@ruuddw OK, thanks. Is there an easy way to configure nSIM to not enable RAM for the full address space?

andrewboie, 2020-02-14:
A workaround has been put in place.
Leaving this open to explore configuration of nSIM to not emulate the entire address space, using DTS as a bounds.

commit 2c9f4210 (2020-02-14): tests: syscalls: exclude one case on nsim_sem

This works around an issue with this emulator's configuration where
there is no memory address that can be poked to generate a fault,
it is simulating memory for the entire address space.

Fixes: #22561
closed 2020-04-10 by andrewboie