zephyrproject-rtos/zephyr #22594: NXP S32K144 MCU support

novalisek, 2020-02-07:
**Is your feature request related to a problem? Please describe.**
I would like to use Zephyr with NXPs S32K144 

**Describe the solution you'd like**
Add support for S32K144 (or the whole S32K family), specially support of all 3 CAN interfaces including CAN FD

**Describe alternatives you've considered**
There is no alternative to this automotive MCU for me now

**Additional context**
Link to S32K   https://www.nxp.com/products/processors-and-microcontrollers/arm-microcontrollers/s32k-automotive-mcus/s32k-scalable-microcontrollers-for-automotive-general-purpose-and-and-high-reliability-industrial:S32K

novalisek, 2024-04-15:
@manuargue would it also be possible to add support for S32K312 (there is also EVB for this   S32K312EVB-Q172 and S32K344 DTS exists, so it should be possible to add DTS for S32K312 )? shall I create new issue for this?

manuargue, 2024-10-15:
NXP currently don't have plans to support S32K1 devices. Community contributions to support other S32K1 devices are welcome. If further questions about NXP plans, please reach out to https://community.nxp.com/

closed 2024-10-15 by manuargue

novalisek, 2024-10-15:
@manuargue Do you know if there any list/roadmap with MCUs from S32K family that should be supported? Thanks.