zephyrproject-rtos/zephyr #22622: tests/drivers/gpio/gpio_basic_api failed on ARM architecture.

chen-png, 2020-02-08:
**Describe the bug**
tests/drivers/gpio/gpio_basic_api failed on reel_board, frdm_k64f, sam_e70_xplained and  mimxrt1050_evk boards.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b reel_board -p auto tests/drivers/gpio/gpio_basic_api/
2. west flash
3. See error

**Screenshots or console output**

****Booting Zephyr OS build zephyr-v2.1.0-2179-g55eb4a11cdb1****
Running test suite gpio_basic_test

starting test - test_gpio_port
Validate device GPIO_0
Check GPIO_0 output 2 connected to input 3
FATAL output pin not wired to input pin? (out high => in low)

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.1
 - Commit ID: 55eb4a11c

pabigot, 2020-02-08:
This is a documentation issue: people running the tests need to know that for the test to work on a given platform there must be a devicetree overlay for that platform in the `boards` directory, and that the two pins identified in that overlay need to be shorted together.  This is currently indicated in the output:

> Check GPIO_0 output 2 connected to input 3
> FATAL output pin not wired to input pin? (out high => in low)

There is no overlay for `reel_board` since what it tests would be handled by `nrf52840_pca10056`.  For `frdm_k64f` the overlay specifies that Arduino D0 and D1 must be shorted.

I'll prepare a README for this test to explain all this, unless somebody else does it first.

chen-png, 2020-02-12:
@pabigot Can we use the testcase.yml to exclude those unsupported boards?

because we use the sanitycheck to run all tests in testing system, if we modify the README file, for each board, when running sanitycheck command, we have to use the sanitycheck's option to exclude this test.
if we can modify the testcase.yml, we can just run sanitycheck command, don't care about boards.

pabigot, 2020-02-12:
@chen-png Probably.  Please see #22754; I think it should exclude unsupported boards.

chen-png, 2020-02-14:
>  Please see #22754; I think it should exclude unsupported boards.

@pabigot  it seems that it only excludes the reel_board,  for frdm_k64f, sam_e70_xplained and mimxrt1050_evk, still the same error.

pabigot, 2020-02-14:
Does it work if you short the input pin to the output pin, as directed in the README?

Because the test simply cannot pass on a board that doesn't have the test configuration.  There's nothing I can do about that.

nashif, 2020-03-11:
@chen-png  please follow up

chen-png, 2020-03-26:
@pabigot  for sam_e70_xplained,  according to the board's overlay, i connected portd 21 and portd 20, then this test passed.
but for frdm_k64f and mimxrt1050_evk, i couldn't find the GPIO pin on board, so do you know where the pin is or have any board manual?

pabigot, 2020-03-26:
For `frdm_k64f` the overlay specifies these are Arduino D0 and D1.   When oriented with the ethernet jack on the front at tht top, these are the lowest two pins on the right side outer column of female headers.   There's silkscreen on the back that shows which pins are on each header.

The same two pins are used on the arduino header of mimxrt10x0_evk.  The header on the 1060 next to the 20-pin JTAG header.

chen-png, 2020-03-31:
@pabigot  thanks for your information.
shorted related pins and tested this case on frdm_k64f and mimxrt1050_evk, this case passed.

PR #22639 (2020-02-12): tests: drivers: gpio_basic_api: add README
Successful execution of this test requires a board-specific overlay to
identify the test pins, and that the test pins be shorted together.
Document this requirement.

Fixes #22622
Fixes #22522
PR #22754 (2020-02-12): tests: gpio_basic_api: fix filter to exclude boards without overlay
The test case no longer permits inferring input and output pins based
only on the presence of GPIO aliases.  Stop allowing presence of GPIO
aliases to enable the test.

Fixes #22622
closed 2020-03-31 by chen-png