zephyrproject-rtos/zephyr #22623: tests/kernel/timer/timer_api failed on mimxrt1050_evk board.

chen-png, 2020-02-08:
**To Reproduce**
Steps to reproduce the behavior:
1.  west build -b mimxrt1050_evk -p auto tests/kernel/timer/timer_api/
2.  west flash
3.  See error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.1.0-2157-g4eaac5c7b379 ***

Running test suite timer_api
starting test - test_time_conversions

PASS - test_time_conversions
starting test - test_timer_duration_period

**Assertion failed at ZEPHYR_BASE/tests/kernel/timer/timer_api/src/main.c:64: duration_expire: interval >= 100 is false**
FAIL - test_timer_duration_period
starting test - test_timer_period_0

PASS - test_timer_period_0
starting test - test_timer_expirefn_null

PASS - test_timer_expirefn_null
starting test - test_timer_periodicity

PASS - test_timer_periodicity
starting test - test_timer_status_get

PASS - test_timer_status_get
starting test - test_timer_status_get_anytime

PASS - test_timer_status_get_anytime
starting test - test_timer_status_sync

PASS - test_timer_status_sync
starting test - test_timer_k_define

PASS - test_timer_k_define
starting test - test_timer_user_data

PASS - test_timer_user_data
starting test - test_timer_remaining_get

PASS - test_timer_remaining_get
Test suite timer_api failed.
PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: Zephyr-sdk-0.11.1
 - Commit ID: 4eaac5c7

MaureenHelm, 2020-02-14:
I was able to reproduce this on 4eaac5c7b379 but not on the latest master d55c8ead03491946d0260d50c9159bb8f8de758f. Bisecting landed on 8153144de0d27fff686647c4d2b2baf08fa462ee
```
commit 8153144de0d27fff686647c4d2b2baf08fa462ee (refs/bisect/new)
Author: Andy Ross <andrew.j.ross@intel.com>
Date:   Thu Feb 6 12:58:53 2020 -0800

    kernel/fatal: Fatal errors must not be preempted
    
    The code underneath z_fatal_error() (which is usually run in an
    exception context, but is not required to be) was running with
    interrupts enabled, which is a little surprising.
    
    The only bug present currently is that the CPU ID extracted for
    logging is subject to a race (i.e. it's possible but very unlikely
    that such a handler might migrate to another CPU after the error is
    flagged and log the wrong CPU ID), but in general users with custom
    error handlers are likely to be surprised when their dying threads
    gets preempted by other code before they can abort.
    
    Signed-off-by: Andy Ross <andrew.j.ross@intel.com>
```

@andyross can you have a look at this? does it make sense that this patch would fix the failure?

andyross, 2020-02-14:
Not really...  Locking can change timing, but off hand I can't even think of a circumstance where a z_fatal_error() is called in a successful run of the timer_api test.  If I had to guess this is just a spurious failure with a failure rate that changes a little with each commit.  Certainly that patch is not going to have fixed anything that was a legitimate timing error.

yingmingx, 2020-02-15:
I think this issue is similar with the problem I've ever try to sovle list below. 
https://github.com/zephyrproject-rtos/zephyr/issues/21813
According to andyross previously suggestion at
https://github.com/zephyrproject-rtos/zephyr/pull/22057
Is there some bug's possibility at low layer driver only on freescale and nxp boards or on all of the boards which system clock is very high?

ioannisg, 2020-02-18:
This test fails for nrF52 platforms as well - might not be the same error though.

pabigot, 2020-02-19:
> This test fails for nrF52 platforms as well - might not be the same error though.

I think it's unrelated, and also not new.  Nordic fails this specific test (line 64) intermittently and has for at least several releases.  This is arguably a failure in the test, which measures in ms durations that are expressed in ticks.

In the failure case the uptime delta is measured as 99 ms when 100 ms is expected.  Since the Nordic system clock advances at 32 KiHz 100 ms cannot be exactly represented: 3276 ticks is 99.976 ms while 3277 ticks is 100.006 ms.  Depending on the tick/ms alignment when the timer was scheduled the difference of the converted values may be 99 or 100 even though the duration in ticks satisfies the interval.

MaureenHelm, 2020-02-24:
This test still passes on master at e4479e2feceee854f4436816c04ee98ad981b66e

closed 2020-02-24 by MaureenHelm