zephyrproject-rtos/zephyr #22624: tests/kernel/semaphore/semaphore failed on iotdk board.

chen-png, 2020-02-08:
**Describe the bug**
Running  tests/kernel/semaphore/semaphore on iotdk board, 
it would stop at "starting test - test_sem_take_timeout_isr" and timeout.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b iotdk -p auto tests/kernel/semaphore/semaphore/
2. west flash
3. see error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.1.0-2157-g4eaac5c7b379 ***

Running test suite test_semaphore
starting test - test_k_sema_init

PASS - test_k_sema_init
starting test - test_sema_thread2thread

PASS - test_sema_thread2thread
starting test - test_sema_thread2isr

PASS - test_sema_thread2isr
starting test - test_sema_reset

PASS - test_sema_reset
starting test - test_sema_count_get

PASS - test_sema_count_get
starting test - test_simple_sem_from_isr

PASS - test_simple_sem_from_isr
starting test - test_simple_sem_from_task

PASS - test_simple_sem_from_task
starting test - test_sem_take_no_wait

PASS - test_sem_take_no_wait
starting test - test_sem_take_no_wait_fails

PASS - test_sem_take_no_wait_fails
starting test - test_sem_take_timeout_fails

PASS - test_sem_take_timeout_fails
starting test - test_sem_take_timeout

PASS - test_sem_take_timeout
starting test - test_sem_take_timeout_forever

PASS - test_sem_take_timeout_forever
**starting test - test_sem_take_timeout_isr**

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.11.1
 - Commit ID: 4eaac5c7b

andrewboie, 2020-02-11:
Possibly the same issue: #22559

vonhust, 2020-02-12:
can be fixed by #22608

jhedberg, 2020-02-12:
@chen-png could you retest with latest master, now that #22608 has been merged?

andrewboie, 2020-02-12:
This should be fixed now.

chen-png, 2020-02-13:
yes, this has been fixed in the latest master with the merged #22608.

closed 2020-02-13 by chen-png