zephyrproject-rtos/zephyr #22625: tests/subsys/canbus/isotp/conformance failed on frdm_k64f board.

chen-png, 2020-02-08:
**To Reproduce**
Steps to reproduce the behavior:
1.  west build -b frdm_k64f -p auto tests/subsys/canbus/isotp/conformance/
2.  west flash
3.  see error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.1.0-2157-g4eaac5c7b379 ***

Running test suite isotp_conformance
starting test - test_send_sf
W: Unhandled error/status (status 0x000014c0, result = 0x00000010

PASS - test_send_sf
starting test - test_receive_sf
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
I: SF DL does not match. Ignore
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
I: SF DL does not match. Ignore
W: Unhandled error/status (status 0x000014c0, result = 0x00000010

PASS - test_receive_sf
starting test - test_send_sf_ext
W: Unhandled error/status (status 0x000014c0, result = 0x00000010

PASS - test_send_sf_ext
starting test - test_receive_sf_ext
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
I: SF DL does not match. Ignore
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
I: SF DL does not match. Ignore
W: Unhandled error/status (status 0x000014c0, result = 0x00000010

PASS - test_receive_sf_ext
starting test - test_send_data
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
W: Unhandled error/status (status 0x000014b9, result = 0x00000000
W: Unhandled error/status (status 0x000014c0, result = 0x00000010

**Assertion failed at ZEPHYR_BASE/tests/subsys/canbus/isotp/conformance/src/main.c:246: check_frame_series: (ret not equal to 0)
Timeout waiting for msg nr 1. ret: -11**

FAIL - test_send_data
starting test - test_send_data_blocks
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
W: Unhandled error/status (status 0x000014b9, result = 0x00000001
W: Unhandled error/status (status 0x000014c0, result = 0x00000010

Assertion failed at ZEPHYR_BASE/tests/subsys/canbus/isotp/conformance/src/main.c:246: check_frame_series: (ret not equal to 0)
Timeout waiting for msg nr 1. ret: -11

FAIL - test_send_data_blocks
starting test - test_receive_data
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
E: Sequence number missmatch
W: Unhandled error/status (status 0x00000001, result = 0x00000004
W: Unhandled error/status (status 0x000014c0, result = 0x00000010
W: Unhandled error/status (status 0x000014c0, result = 0x00000010

**Assertion failed at ZEPHYR_BASE/tests/subsys/canbus/isotp/conformance/src/main.c:246: check_frame_series: (ret not equal to 0)
Timeout waiting for msg nr 0. ret: -11**

FAIL - test_receive_data
starting test - test_receive_data_blocks
E: Error attaching FF filter [-1]
E: Can't attach filter for binding

**Assertion failed at ZEPHYR_BASE/tests/subsys/canbus/isotp/conformance/src/main.c:580: test_receive_data_blocks: (ret not equal to ISOTP_N_OK)
Binding failed [-10]**

FAIL - test_receive_data_blocks
starting test - test_send_timeouts
E: Error attaching FC filter [-1]
E: Can't attach fc filter: -10

**Assertion failed at ZEPHYR_BASE/tests/subsys/canbus/isotp/conformance/src/main.c:625: test_send_timeouts: (ret not equal to ISOTP_N_TIMEOUT_BS)
Expected timeout but got -10**

FAIL - test_send_timeouts
starting test - test_receive_timeouts
E: Error attaching FF filter [-1]
E: Can't attach filter for binding

**Assertion failed at ZEPHYR_BASE/tests/subsys/canbus/isotp/conformance/src/main.c:681: test_receive_timeouts: (ret not equal to ISOTP_N_OK)
Binding failed [-10]**

FAIL - test_receive_timeouts
starting test - test_stmin

**Assertion failed at ZEPHYR_BASE/tests/subsys/canbus/isotp/conformance/src/main.c:272: attach_msgq: (filter_id equal to CAN_NO_FREE_FILTER)
Filter full**

FAIL - test_stmin
starting test - test_receiver_fc_errors

**Assertion failed at ZEPHYR_BASE/tests/subsys/canbus/isotp/conformance/src/main.c:272: attach_msgq: (filter_id equal to CAN_NO_FREE_FILTER)
Filter full**

FAIL - test_receiver_fc_errors
starting test - test_sender_fc_errors

**Assertion failed at ZEPHYR_BASE/tests/subsys/canbus/isotp/conformance/src/main.c:272: attach_msgq: (filter_id equal to CAN_NO_FREE_FILTER)
Filter full**

FAIL - test_sender_fc_errors
Test suite isotp_conformance failed.
PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.11.1
 - Commit ID: 4eaac5c7

alexanderwachter, 2020-02-08:
I don't have a FRDM-K64F board by hand at the moment. I'll check it in a week.

chen-png, 2020-03-18:
this issue caused low pass rate on frdm_k64f board in QA, so change it's priority to high.

nashif, 2020-03-18:
> this issue caused low pass rate on frdm_k64f board in QA, it needs to be fixed, so change it's priority to high.

did this change ever pass in the past?

alexanderwachter, 2020-03-18:
The regression was introduced after the HAL upgrade.
@henrikbrixandersen any findings so fat?

henrikbrixandersen, 2020-03-18:
> The regression was introduced after the HAL update.
> @henrikbrixandersen any findings so far?

No. I have not seen similar issues with CANopen.

chen-png, 2020-03-19:
> did this change ever pass in the past?

this test tests/subsys/canbus/isotp/conformance/ was added into repo since commit "3650f21bd8e", it always failed on frdm_k64f board.

alexanderwachter, 2020-03-19:
If you revert 1e6520c0b06b674190f4a574761dd9c5f4b02142, The test pass. I tested locally before this commit was merged. We need to investigate what has changed there.

nashif, 2020-03-31:
1e6520c0b06b674190f4a574761dd9c5f4b02142 does not exist in the tree...

alexanderwachter, 2020-03-31:
@nashif it is in the nxp hal.

nashif, 2020-03-31:
@alexanderwachter  found it, thanks. @MaureenHelm FYI

carlescufi, 2020-05-21:
@MaureenHelm will you revert the HAL commit that @alexanderwachter linked?

alexanderwachter, 2020-05-21:
The new M_CAN has the same problem. It seems that the issue is in the ISO-TP implementation. I already invested some hours but wasn‘t able to find the bug.

henrikbrixandersen, 2020-05-22:
> @MaureenHelm will you revert the HAL commit that @alexanderwachter linked?

I do not believe that is the correct solution. The HAL update brought in many other fixes.
I believe this issue is within the ISO-TP implementation.

henrikbrixandersen, 2020-05-27:
The ISO-TP sample also fails on the TWR-KE18F board.

PR #25900 (2020-06-02): drivers: can: flexcan: handle rx overflow and undhandled irqs
Add handling of RX overflows and IRQs reported as undhandled by the NXP MCUX HAL.

Fixes: #22625
commit 974e05a9 (2020-06-02): drivers: can: flexcan: handle rx overflow and undhandled irqs

Add handling of RX overflows and IRQs reported as undhandled by the
NXP MCUX HAL.

Fixes: #22625
closed 2020-06-02 by carlescufi