zephyrproject-rtos/zephyr #22652: [Coverity CID :208197] Pointless string comparison in tests/drivers/gpio/gpio_basic_api/src/main.c

zephyrbot, 2020-02-09:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/e089906b339aad4cd1b6589a3b6ce94782d93f54/tests/drivers/gpio/gpio_basic_api/src/main.c#L27

Category: Incorrect expression
Function: `board_setup`
Component: Tests
CID: [208197](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=208197)

Details:

```
21     #endif
22    
23     static void board_setup(void)
24     {
25     #ifdef DT_INST_0_TEST_GPIO_BASIC_API
26         /* PIN_IN and PIN_OUT must be on same controller. */
>>>     CID 208197:  Incorrect expression  (CONSTANT_EXPRESSION_RESULT)
>>>     "strcmp("GPIO_A2", "GPIO_A2")" is always 0 because ""GPIO_A2"" is compared against itself.
27         if (strcmp(DT_INST_0_TEST_GPIO_BASIC_API_OUT_GPIOS_CONTROLLER,
28                DT_INST_0_TEST_GPIO_BASIC_API_IN_GPIOS_CONTROLLER) != 0) {
29             printk("FATAL: output controller %s != input controller %s\n",
30                    DT_INST_0_TEST_GPIO_BASIC_API_OUT_GPIOS_CONTROLLER,
31                    DT_INST_0_TEST_GPIO_BASIC_API_IN_GPIOS_CONTROLLER);
32             k_panic();
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

pabigot, 2020-02-09:
Coverity needs to be told to ignore this.  I don't have authority to tell it.

pabigot, 2020-02-11:
Marked false-positive, ignore in coverity.

closed 2020-02-11 by pabigot