zephyrproject-rtos/zephyr #22653: [Coverity CID :208196] Out-of-bounds access in drivers/eeprom/eeprom_simulator.c

zephyrbot, 2020-02-09:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/e089906b339aad4cd1b6589a3b6ce94782d93f54/drivers/eeprom/eeprom_simulator.c#L259

Category: Memory - corruptions
Function: `eeprom_sim_init`
Component: Drivers
CID: [208196](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=208196)

Details:

```
254     #endif /* CONFIG_ARCH_POSIX */
255    
256     static int eeprom_sim_init(struct device *dev)
257     {
258         SYNC_INIT();
259         STATS_INIT_AND_REG(eeprom_sim_stats, STATS_SIZE_32, "eeprom_sim_stats");
>>>     CID 208196:    (OVERRUN)
>>>     Overrunning struct type stats_hdr of 24 bytes by passing it to a function which accesses it at byte offset 24.
260         STATS_INIT_AND_REG(eeprom_sim_thresholds, STATS_SIZE_32,
261                    "eeprom_sim_thresholds");
262    
263         return eeprom_mock_init(dev);
264     }
265    
253    
254     #endif /* CONFIG_ARCH_POSIX */
255    
256     static int eeprom_sim_init(struct device *dev)
257     {
258         SYNC_INIT();
>>>     CID 208196:    (OVERRUN)
>>>     Overrunning struct type stats_hdr of 24 bytes by passing it to a function which accesses it at byte offset 24.
259         STATS_INIT_AND_REG(eeprom_sim_stats, STATS_SIZE_32, "eeprom_sim_stats");
260         STATS_INIT_AND_REG(eeprom_sim_thresholds, STATS_SIZE_32,
261                    "eeprom_sim_thresholds");
262    
263         return eeprom_mock_init(dev);
264     }
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

henrikbrixandersen, 2020-02-10:
@Laczen Any idea why Coverity has flagged this?

pabigot, 2020-02-10:
I would expect that Coverity is unaware of the linker magic that tries to put the stats elements immediately after the header.  At a language level there's no guarantee that the bytes after it are part of the same object.

Moving to a flexable array member solution like:

```
struct stats_hdr {
        const char *s_name;
        u8_t s_size;
        u16_t s_cnt;
        u8_t s_pad1;
#ifdef CONFIG_STATS_NAMES
        const struct stats_name_map *s_map;
        int s_map_cnt;
#endif
        struct stats_hdr *s_next;
	union {
		u8_t u8[];
		u16_t u16[];
		u32_t u32[];
	} elts;
};
```

might allow for an implementation that avoids the diagnostic.

Laczen, 2020-02-10:
@henrikbrixandersen, no I have no idea. The usage of stats was copied from flash_simulator.

Laczen, 2020-02-11:
@nashif, as the author of the stats functionality you might be the best placed to comment.

nvlsianpu, 2020-02-13:
@henrikbrixandersen This is false-positive.
@pabigot Interesting suggestion.

galak, 2020-02-19:
Marked it in coverity as False Positive.

closed 2020-02-19 by galak

pabigot, 2020-02-19:
I personally don't consider this a "false positive": the code truly is walking past the end of the object that's pointed to, and there's a language-sanctioned way to indicate that doing so is safe that could be used instead.

@MaureenHelm @dleach02 @nashif I think you've all expressed positions on the question informally.  Is it the security or functional safety group that should be given the responsibility of defining the policy on handling Coverity diagnostics?