zephyrproject-rtos/zephyr #22656: [Coverity CID :208192] Out-of-bounds access in tests/subsys/canbus/isotp/implementation/src/main.c

zephyrbot, 2020-02-09:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/e089906b339aad4cd1b6589a3b6ce94782d93f54/tests/subsys/canbus/isotp/implementation/src/main.c#L197

Category: Memory - corruptions
Function: `receive_test_data`
Component: Tests
CID: [208192](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=208192)

Details:

```
196             recv_len = isotp_recv(recv_ctx, data_buf, sizeof(data_buf),
197                           K_MSEC(1000));
198             zassert_true(recv_len >= 0, "recv error: %d", recv_len);
199    
200             zassert_true(remaining_len >= recv_len,
201                      "More data then expected");
>>>     CID 208192:    (OVERRUN)
>>>     Overrunning array "data_buf" of 128 bytes by passing it to a function which accesses it at byte offset 4294967286 using argument "recv_len" (which evaluates to 4294967287).
202             ret = check_data(data_buf, data_ptr, recv_len);
203             zassert_equal(ret, 0, "Data differ");
204             data_ptr += recv_len;
205             remaining_len -= recv_len;
206    
207             if (delay) {
191             memset(data_buf, 0, sizeof(data_buf));
192             ret = isotp_recv(recv_ctx, data_buf, sizeof(data_buf),
193                      K_MSEC(1000));
194             zassert_true(data_buf >= 0, "recv error: %d", ret);
195    
196             zassert_true(remaining_len >= ret, "More data then expected");
>>>     CID 208192:    (OVERRUN)
>>>     Overrunning array "data_buf" of 128 bytes by passing it to a function which accesses it at byte offset 4294967286 using argument "ret" (which evaluates to 4294967287).
197             check_data(data_buf, data_ptr, ret);
198             data_ptr += ret;
199             remaining_len -= ret;
200    
201             if (delay) {
202                 k_sleep(delay);
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

closed 2020-02-10 by jukkar

ceolin, 2020-03-06:
@alexanderwachter @jukkar coverity is still reporting this issue, AFAIU it is a false positive, coverity seems to get confuse with zassert. Can you just confirm that  and mark as false positive there ? If you don't have permission, just let me know that I do it.

alexanderwachter, 2020-03-07:
Marked as a false positive.

PR #22661 (2020-02-10): canbus: isotp: Fix nullptr deref and wrong variable in assert (Coverity issues)
This PR is a fix for CID 208191(#22657) and CID 208192 (#22656)

fixes #22657 
fixes #22656
commit 1faaff1e (2020-02-10): tests: canbus: isotp: Fix wrong variable used for assert

The assert for error checking used the wrong variable.
data_buf was checked instead of the ret variable.

Fixes #22656
Coverity-CID: 208192