zephyrproject-rtos/zephyr #22702: Implement I2S driver for lpcxpresso55s69

MaureenHelm, 2020-02-10:
**Is your enhancement proposal related to a problem? Please describe.**
The `lpcxpresso55s69` board currently lacks support for I2S in zephyr.

**Describe the solution you'd like**
Implement a shim driver that adapts the mcux lpc I2S (flexcomm) driver to the zephyr I2S interface. Enable `tests/drivers/i2s` for this board to ensure the I2S driver builds and functions correctly.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

tobyfirth, 2020-08-14:
I have been having a look at the I2S Zephyr api and tests and the LPC I2S api and the lpc55s6x user manual. From my understanding of the Zephyr API and how it is used in the tests, it looks like the API assumes that an I2S instance can operate as a sender and a receive at the same time. The LPC API and the user manual seem to be quite clear that an I2S instance can only operate as a sender or a receiver and not both at the same time.

Assuming I have understood the Zephyr API and the LPC API properly, I can think of the following possible options;
1. Pretend that the LPC I2S can do send and receive and hide that it can't to pass the tests.
2. Have the driver being able to swap between send and receive depending on the API called.
3. Change the tests so they don't require to be able to send and receive at the same time, at least not for all I2S drivers.
4. Change the I2S API acknowledge that some I2S devices that can send and receive and that some can't.

mmahadevan108, 2021-09-10:
This support is now available in the main branch of Zephyr after this PR has been merged: https://github.com/zephyrproject-rtos/zephyr/pull/38072

mmahadevan108, 2021-09-10:
Can this issue be resolved

closed 2021-09-14 by MaureenHelm