zephyrproject-rtos/zephyr #22704: Implement watchdog driver for lpcxpresso55s69

MaureenHelm, 2020-02-10:
**Is your enhancement proposal related to a problem? Please describe.**
The `lpcxpresso55s69` board currently lacks support for watchdog in zephyr.

**Describe the solution you'd like**
Implement a shim driver that adapts the mcux lpc wwdt driver to the zephyr watchdog interface. The existing `wdt_mcux_wdog.c` and `wdt_mcux_wdog32.c` shim drivers for kinetis and i.mx rt can be used for reference. Enable `tests/drivers/watchdog` for this board to ensure the watchdog driver builds and functions correctly.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Nephirus, 2020-07-09:
Here is PR for NXP HAL, that builds relevant HAL driver when shim driver is enabled: https://github.com/zephyrproject-rtos/hal_nxp/pull/48

I also have a working shim driver for the watchdog, I'll create PR for that soon (after some clean-up).

MaureenHelm, 2020-07-09:
Great, thank you!

Nephirus, 2020-07-09:
No problem :)

Promised PR for shim driver: https://github.com/zephyrproject-rtos/zephyr/pull/26779

`samples/drivers/watchdog` works on LPCXpresso55S69 board, but I'm not 100% sure about Kconfig related stuff.

commit 3a87517c (2020-08-04): drivers: watchdog: mcux_wwdt: add driver for NXP LPC 55xxx WWDT

Adds shim driver for LPC 55xxx Windowed Watchdog Timer (WWDT).

Also updates the "Supported features" table in LPCXpresso55S69
board documentation page.

Fixes #22704.
closed 2020-08-04 by MaureenHelm