zephyrproject-rtos/zephyr #22705: Implement counter driver for lpcxpresso55s69

MaureenHelm, 2020-02-10:
**Is your enhancement proposal related to a problem? Please describe.**
The `lpcxpresso55s69` board currently lacks support for counter in zephyr.

**Describe the solution you'd like**
Implement a shim driver that adapts the mcux lpc RTC driver to the zephyr counter interface. The existing `counter_mcux_rtc.c` and `counter_mcux_gpt.c` shim drivers for kinetis and i.mx rt can be used for reference. Enable `tests/drivers/counter` for this board to ensure the counter driver builds and functions correctly.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

puranjaymohan, 2020-05-05:
Hi, I want to work on this as a first-time contributor. Please tell me if a contribution wiki or guide is available?

galak, 2020-05-05:
https://docs.zephyrproject.org/latest/contribute/index.html

PR #37613 (2021-08-24): drivers: counter: added ctimer driver for lpcexpresso55s69
Added shim driver for the CTIMERs for the lpcexpresso55s69 board.

Fixes: #22705
commit 893bfc0f (2021-08-24): drivers: counter: added ctimer driver for lpcexpresso55s69

Added shim driver for the CTIMERs for the lpcexpresso55s69 board.

Fixes: #22705
closed 2021-08-24 by nashif